Aquantia/AQtion #1: Kernel Panic & memory problems

tbludau, 2018-04-09:
Hello,

i've identified on all my systems with the Asus XG-C100C PCIe/XG network card and the atlantic driver a lot of strange problems under Ubuntu 16.04 and 18.04 (lts) with the newest kernel versions. ( 4.15.15-041515-generic, 4.15.16-041516-generic, linux-image-4.15.0-13-generic)

So i've updated to the newest driver 2.0.6. but the problems are still there.

After a short time when some traffic is comming over the network i got a lot of messages with: "general protection fault: 0000 [#3] SMP NOPTI" with a call trace and randomly a lot of programms and modules  (samba, ext4, network, etc.) crashes and randomly the kernel crashes too. I searched a lot of time where this problem is coming from and identified the atlantic driver as source of the evil.

Under Windows (dual boot system) everything works normal.
[modinfo-atlantic.txt](https://github.com/Aquantia/AQtion/files/1889988/modinfo-atlantic.txt)

[crash-4.15.15-041515-generic.txt](https://github.com/Aquantia/AQtion/files/1889983/crash-4.15.15-041515-generic.txt)
[crash-4.15.16-041516-generic.txt](https://github.com/Aquantia/AQtion/files/1889986/crash-4.15.16-041516-generic.txt)

On RedHat they identified a equivalent problem some month before.

When you give me some hints what you need for debugging i will send you everything.

The Sytems are:
AMD Turion(tm) II Neo N54L Dual-Core Processor
and
AMD FX(tm)-8120 Eight-Core Processor

tbludau, 2018-04-09:
Link from RedHat: https://bugzilla.redhat.com/show_bug.cgi?id=1499321

cail, 2018-04-10:
Thanks for reporting this!
Could you please give us you `sudo hwinfo` output?

tbludau, 2018-04-10:
I send you the hwinfo over mail.

At the moment i've created a stable enviornment.

The problems i've described were definitly a problem with the drivers from the delivered kernel version (1.6.13.0-kern). And after some compiling and tests i think i forgot to integrate the module in the initrd and used the old one.

So this is the second time i have this stable state. 
At the moment everything looks as expected. (with a bug of the link detection on the N54L System and driver 2.0.6, but thats another problem)

cail, 2018-04-11:
So... 2.0.6 which is here has no this issue visible?

tbludau, 2018-04-11:
After some Network-Traffic tests i never seen this issue again with 2.0.6.

So this can be closed too.

closed 2018-04-11 by cail