Aquantia/AQtion #4: Packets dropped

ldhasty, 2019-04-06:
While using this version of the driver I noticed there was a high number of packets dropped using Fedora 29.  I was transferring about 70 gig worth of data via SMB and noticed over 429k packets were dropped before I ended the transfer and reverted to the kernel's driver implementation.  This is using kernel 5.0.5-200.fc29.x86_64 and an Asus XG-C100C(Atlantic 107).  Reverting to the kernels implementation I lost about 1 Gbps in through put but there were no dropped packets

aenertia, 2019-04-07:
Interesting I noticed that on f29 to ubuntu bionic I was getting autoneg failures and when it did negotiate it would only do so at 100mb - forcing the speed to 10000 fixed when I upgraded the ubuntu side kernel to a mainline 4.20.17 kernel. 

Previously ubuntu to ubuntu was working without issue. what firmware does ethtool report for you on the card?

ldhasty, 2019-04-07:
driver: atlantic
version: 2.0.4.0-kern
firmware-version: 3.0.33
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no

cail, 2019-04-08:
@ldhasty, how did you checked the number of drops?
Could you provide the following output before and after the transfer In good and bad cases:
`ethtool -S enp1s0`

ldhasty, 2019-04-08:
Stock Driver 

enp1s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9000
        inet 192.168.1.2  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::5af8:ff3:8911:ef45  prefixlen 64  scopeid 0x20<link>
        ether 40:b0:76:58:43:89  txqueuelen 1000  (Ethernet)
        RX packets 110330575  bytes 49335076640 (45.9 GiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 177051049  bytes 1546257089615 (1.4 TiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Latest Driver

enp1s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9000
        inet 192.168.1.2  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::5af8:ff3:8911:ef45  prefixlen 64  scopeid 0x20<link>
        ether 40:b0:76:58:43:89  txqueuelen 1000  (Ethernet)
        RX packets 1277528  bytes 11439203693 (10.6 GiB)
        RX errors 0  dropped 331707  overruns 0  frame 0
        TX packets 331708  bytes 20262689 (19.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

cail, 2019-04-18:
in kernel driver did not reported "dropped" counter. Thats why you don't see it increasing.
I guess its always increasing, in both drivers.
You may check in kernel driver ethtool -S, line 
InDroppedDma: 656988
will show you actual drops. Guess it'll be non zero.

Drops are inevitable on high rates, especially with udp-based protocols.