Aquantia/AQtion #5: Bridged with 1Gb Interface - Connection drops

CMDRZOD, 2019-04-17:
I built a router based on Ubuntu 18.04. I've got 3 NIC's in this router PC. Asus XG-C100C, intel integrated 1Gb plus a 1 port Intel NIC in a PCIE slot.

The integrated NIC is plugged into my fibre modem, and the Asus and Intel PCIE cards are bridged together. This works pretty well, but I often get random disconnects on the Asus XG 10Gb card. It is plugged into my Windows 10 PC which also has an Asus XG-C100C NIC to facilitate 10Gb transfers between my windows PC and the Ubuntu router.

I've disabled LRO and GRO, enabled STP and set MTU to 9000 for all bridges and their interfaces. None of these have made any difference. Last night I discovered this driver and decided to try it instead of the driver available on the Aquantia website. Both drivers exhibit this issue.

Any hints on troubleshooting and narrowing down the problem would be appreciated. Thanks for all your hard work this driver appears to get slightly better throughput than the Aquantia provided driver. Attached is the output of journalctl when this disconnect happens. 

> Apr 17 09:46:02 kknd systemd-networkd[974]: enp1s0: Lost carrier
Apr 17 09:46:02 kknd systemd-timesyncd[832]: Network configuration changed, trying to establish connection.
Apr 17 09:46:02 kknd kernel: atlantic 0000:01:00.0 enp1s0: atlantic: link change old 10000 new 0
Apr 17 09:46:02 kknd kernel: br0: port 2(enp1s0) entered disabled state

I've also made a post on reddit, here is a link to that as well, may have some useful information.
[https://www.reddit.com/r/HomeNetworking/comments/be1boe/ubuntu_server_1804_router_bridged_aqn107/](url)

cail, 2019-04-18:
Since both 10G NICs are interconnected, it could be a line issue on each of them, not exactly on ubuntu side.
Do you see link is getting restored after that link down? How fast if so? Does cable plug/unplug helps?

CMDRZOD, 2019-04-18:
I can almost always restart the link by:
Disable/Enable the Windows 10 Asus XG-C100C
Unplugging and replugging the cable will work as well.

If I just try ipconfig /renew /release on the Windows machine the prompt will hang forever.

Everytime the link is up it is negotiated at 10gb. I build a new cat6 last night with no difference.

I also plugged from my router 10gb card to the Windows 10 PC Intel integrated 1Gb NIC. No connection drops while in 1Gb mode. 

I plugged back into the Asus card for 10gb link, and got a disconnect right away.

When this disconnect happens, the windows troubleshooter will usually say gateway lost as well.

CMDRZOD, 2019-04-28:
Just updated both of my AQN-107's to the latest firmware posted on station-drivers. I used Windows 10 to do it. Put them both back in and will see if this fixes the issue. New ethtool -i:

driver: atlantic
version: 2.2.3.0
firmware-version: 3.1.56
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: yes

CMDRZOD, 2019-05-03:
After a few days of using the newer firmware, the issue seems to be much less frequent, but still present. About every 24-48 hours the NIC in the router goes unresponsive. Doing more research today I noticed this in my ifconfig:

enp1s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
ether 2c:fd:a1:dc:f8:d4  txqueuelen 1000  (Ethernet)
RX packets 30407  bytes 8196102 (8.1 MB)
RX errors 0  dropped 0  overruns 0  frame 0
TX packets 37270  bytes 23422424 (23.4 MB)
TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Notice it does not say RUNNING. The other interface in the bridge as well as the bridge itself both report running still, but the Asus 10Gb NIC shown above does not.

What can I do to further troubleshoot this issue? The only thing that seems to work now is a restart of the router PC. I also uncovered these messages during boot:

May 02 00:06:09 kknd kernel: pcieport 0000:00:1c.4: AER: Corrected error received: 0000:00:1c.4
May 02 00:06:09 kknd kernel: pcieport 0000:00:1c.4: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Receiver ID)
May 02 00:06:09 kknd kernel: pcieport 0000:00:1c.4:   device [8086:a294] error status/mask=00000001/00002000
May 02 00:06:09 kknd kernel: pcieport 0000:00:1c.4:    [ 0] Receiver Error         (First)

Now that I have noticed this error that is present with every boot, I will try to run the Ubuntu router PC with the boot option pci=noaer.

If there is no improvement I will try pci=nomsi. If anyone else has any suggestions I'd love to hear them.

CMDRZOD, 2019-05-03:
Didn't mean to close it sorry.

cail, 2019-05-06:
Can you give more details on your router PC hardware?
Please attach output of `sudo lspci -vvvvv`
After bad thing happens, could you please try doing ifconfig enp1s0 down, then ifconfig enp1s0 up - and see if that restores the link?

CMDRZOD, 2019-05-06:
I just tried ifconfig enp1s0 down then up, it did restore the link. When I got home today the link was down but this successfully recovered the connection. 

The router PC is a ROG Strix Z370-Gaming-Wifi-AC Motherboard. It has a i3 8350k intel CPU with 32GB of RAM. In the PCI-E I have the Asus XG-C100C 10Gb NIC as well as an intel 1Gb NIC. Attached is the output you requested. Also I have set pci=noaer in an attempt to fix the intermittent connection drop. This appears to have no effect, except to suppress error messages from the PCI bus during boot.

00:00.0 Host bridge: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
	Subsystem: ASUSTeK Computer Inc. Device 8694
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ >SERR- <PERR- INTx-
	Latency: 0
	Capabilities: [e0] Vendor Specific Information: Len=10 <?>

00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 08) (prog-if 00 [Normal decode])
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 120
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 0000f000-00000fff
	Memory behind bridge: f7000000-f74fffff
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [88] Subsystem: ASUSTeK Computer Inc. Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16)
	Capabilities: [80] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [90] MSI: Enable+ Count=1/1 Maskable- 64bit-
		Address: fee04004  Data: 4021
	Capabilities: [a0] Express (v2) Root Port (Slot+), MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 256 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #2, Speed 8GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <256ns, L1 <8us
			ClockPM- Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x4, TrErr- Train- SlotClk+ DLActive- BWMgmt+ ABWMgmt+
		SltCap:	AttnBtn- PwrCtrl- MRL- AttnInd- PwrInd- HotPlug- Surprise-
			Slot #1, PowerLimit 75.000W; Interlock- NoCompl+
		SltCtl:	Enable: AttnBtn- PwrFlt- MRL- PresDet- CmdCplt- HPIrq- LinkChg-
			Control: AttnInd Unknown, PwrInd Unknown, Power- Interlock-
		SltSta:	Status: AttnBtn- PowerFlt- MRL- CmdCplt- PresDet+ Interlock-
			Changed: MRL- PresDet+ LinkState-
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible-
		RootCap: CRSVisible-
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR+, OBFF Via WAKE# ARIFwd-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Via WAKE# ARIFwd-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+, EqualizationPhase1+
			 EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
	Capabilities: [100 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed+ WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [140 v1] Root Complex Link
		Desc:	PortNumber=02 ComponentID=01 EltType=Config
		Link0:	Desc:	TargetPort=00 TargetComponent=01 AssocRCRB- LinkType=MemMapped LinkValid+
			Addr:	00000000fed19000
[... 531 lines cut ...]
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x2, ASPM L0s L1, Exit Latency L0s <2us, L1 unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x2, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR+, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+, EqualizationPhase1+
			 EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
[... cut ...]
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap- ChkEn-
	Capabilities: [200 v1] #19
	Capabilities: [300 v1] Latency Tolerance Reporting
		Max snoop latency: 3145728ns
		Max no snoop latency: 3145728ns
	Kernel driver in use: xhci_hcd

T-Bonhagen, 2022-10-31:
I have the same problem without Bridge:

Linux 5.10.0-18-amd64 #1 SMP Debian 5.10.140-1 (2022-09-02) x86_64 GNU/Linux

[ 1276.025828] atlantic 0000:01:00.0 eth0: atlantic: link change old 10000 new 0
[ 1365.563675] atlantic 0000:01:00.0 eth0: atlantic: link change old 0 new 10000

driver: atlantic
version: 5.10.0-18-amd64
firmware-version: 3.1.100
expansion-rom-version: 
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: yes

00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Root Complex
        Subsystem: ASUSTeK Computer Inc. PRIME B450M-A Motherboard
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-

00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-

00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0] (prog-if 00 [Normal decode])
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin ? routed to IRQ 25
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 0000f000-00000fff [disabled]
        Memory behind bridge: fc000000-fc4fffff [size=5M]
        Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16+ MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: [50] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [58] Express (v2) Root Port (Slot+), MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0
                        ExtTag+ RBE+
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 512 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x8, ASPM L1, Exit Latency L1 <64us
                        ClockPM- Surprise- LLActRep+ BwNot+ ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 8GT/s (ok), Width x4 (downgraded)
                        TrErr- Train- SlotClk+ DLActive+ BWMgmt+ ABWMgmt-
                SltCap: AttnBtn- PwrCtrl- MRL- AttnInd- PwrInd- HotPlug- Surprise-
                        Slot #0, PowerLimit 0.000W; Interlock- NoCompl+
                SltCtl: Enable: AttnBtn- PwrFlt- MRL- PresDet- CmdCplt- HPIrq- LinkChg-
                        Control: AttnInd Unknown, PwrInd Unknown, Power- Interlock-
                SltSta: Status: AttnBtn- PowerFlt- MRL- CmdCplt- PresDet+ Interlock-
                        Changed: MRL- PresDet- LinkState+
                RootCap: CRSVisible+
                RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
                RootSta: PME ReqID 0000, PMEStatus- PMEPending-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix+, MaxEETLPPrefixes 1
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd+
                         AtomicOpsCap: Routing- 32bit+ 64bit+ 128bitCAS-
                DevCtl2: Completion Timeout: 65ms to 210ms, TimeoutDis- LTR- OBFF Disabled, ARIFwd-
[... 1121 lines cut ...]
                LnkSta: Speed 8GT/s (ok), Width x16 (ok)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
                LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
[... cut ...]
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [a0] MSI: Enable+ Count=1/2 Maskable- 64bit+
                Address: 00000000fee00000  Data: 4024
        Capabilities: [d0] SATA HBA v1.0 InCfgSpace
        Capabilities: [100 v1] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [270 v1] Secondary PCI Express
                LnkCtl3: LnkEquIntrruptEn- PerformEqu-
                LaneErrStat: 0
        Kernel driver in use: ahci
        Kernel modules: ahci

ioquatix, 2023-10-01:
I might be having similar problems:

```
Sep 14 09:40:57 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 14 09:40:57 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 14 09:41:06 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 14 09:42:58 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 14 09:43:06 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 15 21:55:16 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 15 21:55:16 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 15 21:55:25 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 15 21:57:17 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 15 21:57:25 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 16 10:56:11 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 16 10:56:11 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 16 10:56:18 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 16 10:57:55 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 16 10:58:03 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 18 20:14:01 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 18 20:14:01 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 18 20:14:10 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 18 20:16:02 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 18 20:16:10 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 20 08:58:00 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 20 08:58:00 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 20 08:58:09 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 20 09:00:02 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 20 09:00:11 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 21 08:45:10 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 21 08:45:10 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 21 08:45:18 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 21 08:46:53 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 21 08:47:01 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 22 08:01:54 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 22 08:01:54 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 22 08:02:01 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 22 08:03:37 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 22 08:03:45 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 22 22:01:10 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 22 22:01:10 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 22 22:01:18 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 22 22:02:53 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 22 22:03:01 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 22 22:23:21 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 22 22:23:22 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 22 22:23:30 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 22 22:25:05 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 22 22:25:12 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 23 10:34:20 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 23 10:34:20 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 23 10:34:28 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 23 10:36:02 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 23 10:36:09 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 24 11:59:24 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 24 11:59:24 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 24 11:59:33 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 24 12:01:26 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 24 12:01:34 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 25 11:56:51 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 25 11:56:51 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 25 11:57:00 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 25 11:58:52 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Sep 25 11:59:00 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 26 18:42:27 aiko kernel: atlantic: Detect ATL2FW 1030018
Sep 26 18:42:27 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
Sep 26 18:42:35 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Sep 26 18:44:10 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
[... cut ...]
Oct 02 10:12:28 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
Oct 02 10:14:02 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
Oct 02 10:14:10 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
```

```
> sudo ethtool -i eno2
driver: atlantic
version: 6.5.4-arch2-1
firmware-version: 1.3.24
expansion-rom-version: 
bus-info: 0000:0a:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes
```

Should I try to update the firmware?

kerem-ak1, 2024-08-19:
> I might be having similar problems:
> 
> ```
> Sep 14 09:40:57 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 14 09:40:57 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 14 09:41:06 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 14 09:42:58 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 14 09:43:06 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 15 21:55:16 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 15 21:55:16 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 15 21:55:25 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 15 21:57:17 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 15 21:57:25 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 16 10:56:11 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 16 10:56:11 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 16 10:56:18 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 16 10:57:55 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 16 10:58:03 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 18 20:14:01 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 18 20:14:01 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 18 20:14:10 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 18 20:16:02 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 18 20:16:10 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 20 08:58:00 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 20 08:58:00 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 20 08:58:09 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 20 09:00:02 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 20 09:00:11 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 21 08:45:10 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 21 08:45:10 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 21 08:45:18 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 21 08:46:53 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 21 08:47:01 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 22 08:01:54 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 22 08:01:54 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 22 08:02:01 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 22 08:03:37 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 22 08:03:45 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 22 22:01:10 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 22 22:01:10 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 22 22:01:18 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 22 22:02:53 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 22 22:03:01 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 22 22:23:21 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 22 22:23:22 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 22 22:23:30 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 22 22:25:05 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 22 22:25:12 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 23 10:34:20 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 23 10:34:20 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 23 10:34:28 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 23 10:36:02 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 23 10:36:09 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 24 11:59:24 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 24 11:59:24 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 24 11:59:33 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 24 12:01:26 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 24 12:01:34 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 25 11:56:51 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 25 11:56:51 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 25 11:57:00 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 25 11:58:52 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 10000 new 0
> Sep 25 11:59:00 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
> Sep 26 18:42:27 aiko kernel: atlantic: Detect ATL2FW 1030018
> Sep 26 18:42:27 aiko kernel: atlantic 0000:0a:00.0 eno2: renamed from eth0
> Sep 26 18:42:35 aiko kernel: atlantic 0000:0a:00.0 eno2: atlantic: link change old 0 new 10000
[... cut ...]
you have rev03,  asus website says if you rev03 you dont need to install firmware upgrade. 

Problem still persist with:

driver: atlantic
version: 6.10.5-arch1-1
firmware-version: 1.3.18
expansion-rom-version: 
bus-info: 0000:05:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes

lspci | grep Aquantia
Aquantia Corp. AQtion AQC113CS NBase-T/IEEE 802.3an Ethernet Controller [Antigua 10G] (rev 03)
Linux arch128 6.10.5-arch1-1