Aquantia/AQtion #8: Missing FW blobs

t-nelson, 2019-09-19:
Hi!  Congratulations on being acquired by Marvell?!  Hopefully that won't affect this repository...

I'm wondering where these FW blobs are hosted?

https://github.com/Aquantia/AQtion/blob/e024a859ea828ced78db310328db90dbedea837a/aq_common.h#L56-L58

Thanks!

cail, 2019-09-20:
Thanks alot! ;)
These blobs are not yet available on public, basically this is not required for the major of end user devices since they comes up with FW flashed always.

t-nelson, 2019-09-20:
Thanks for the quick reply!

Interesting.  I'm seeing
```
kernel: atlantic 0000:03:00.0: Direct firmware load for aquantia/87B1.fw failed with error -2
```
when trying to test load the 2.2.6 release with `insmod ./atlantic.ko` for an AQC107 that's built onto an [ASRock Fatal1ty X399](https://www.asrock.com/mb/AMD/Fatal1ty%20X399%20Professional%20Gaming/index.asp#Specification).  It's running Ubuntu 18.04, with the same behavior on both the standard 4.15 kernel and 5.0 from the `hwe-18.04-edge` kernel package.  I've tried building from both this repository and the tarball that had been hosted on aquantia.com up until yesterday.

The actual issue I'm running from is the device hanging up during high load to the point of needing to reboot the machine to get it working again.  Similar to https://bugzilla.redhat.com/show_bug.cgi?id=1499321, I believe.  It may be the case that this is recoverable with direct terminal access, but these machines are remote and connected to the internet via the AQC107, so I have limited facilities at my disposal ;).  Upon reading the changelogs, it seems as if this is likely resolved in newer driver releases.

Any advice on how to proceed would be greatly appreciated!

cail, 2019-09-20:
That particular message could be ignored, we've fixed it and will release the update soon.

On your hangups, it leaks information to analyse. the issue your are referencing is very old and most probably irrelevant.

Its not clear, do you still see these hangups even with 2.2.6 driver? If you have not tried it, it worth trying. as it contains alot of fixes.

t-nelson, 2019-09-20:
Oh!  I was under the impression that message implied that the module failed to load.  Like I said, the NIC is my only connection to these machines, so I have been very conservative in my testing.  I'd used the following script:
```
$ cat chicken.sh 
#!/usr/bin/env bash

sudo rmmod atlantic
sudo modprobe ptp
sudo modprobe crc_itu_t
sudo insmod ./atlantic.ko
modinfo atlantic
sleep 60
sudo rmmod atlantic
sudo modprobe atlantic
```
I lose connectivity during the sleep and modinfo reports the in-kernel driver version.  The only message in syslog is the firmware load failure.  Confusingly, testing the return codes of `rmmod` and `insmod` both indicate success.

I'll look into it further under this new light and report back

Shekelme, 2022-05-30:
Today I also sow the messages about these files:

cail, 2022-05-30:
These are warnings, no impact on functionality.

closed 2022-05-30 by cail