Aquantia/AQtion #11: Driver does not work on Ubuntu 18.04.3 LTS

themightyoarfish, 2020-01-04:
```
sudo apt-get install ./atlantic_1578154949.deb 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'atlantic' instead of './atlantic_1578154949.deb'
The following packages were automatically installed and are no longer required:
  cmake-data gsfonts-x11 libappindicator1 libaudclient2 libglade2-0 libgsoap-2.8.60 libid3tag0 libimlib2 libindicator7 libllvm7
  libllvm7:i386 liblua5.1-0 libmotif-common librhash0 libuv1 libvncserver1 libxm4 libxmmsclient6 python-appindicator python-gconf
  python-glade2 python-gobject python-gobject-2 python-gtk2 python-pexpect python-ptyprocess python-xdg
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed
  atlantic
0 to upgrade, 1 to newly install, 0 to remove and 19 not to upgrade.
Need to get 0 B/95.6 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/rasmus/Downloads/AQtion/atlantic_1578154949.deb atlantic all 1578154949 [95.6 kB]
Selecting previously unselected package atlantic.
(Reading database ... 372982 files and directories currently installed.)
Preparing to unpack .../AQtion/atlantic_1578154949.deb ...
rmmod: ERROR: Module atlantic is not currently loaded
Unpacking atlantic (1578154949) ...
Setting up atlantic (1578154949) ...
make -j4 -C /lib/modules/4.15.0-72-generic/build M=/var/build_aq_drv/1578154949/Linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-72-generic'
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_main.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_nic.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_pci_func.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_vec.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_ring.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_ptp.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_filters.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_hw_utils.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_ethtool.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_drvinfo.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/hw_atl/hw_atl_a0.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/hw_atl/hw_atl_b0.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/hw_atl/hw_atl_utils.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/hw_atl/hw_atl_utils_fw2x.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_trace.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_phy.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/hw_atl/hw_atl_llh.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/hw_atl/hw_atl_fw_hostboot.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/hw_atl/hw_atl_fw_image.o
  CC [M]  /var/build_aq_drv/1578154949/Linux/aq_compat.o
  LD [M]  /var/build_aq_drv/1578154949/Linux/atlantic.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /var/build_aq_drv/1578154949/Linux/atlantic.mod.o
  LD [M]  /var/build_aq_drv/1578154949/Linux/atlantic.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-72-generic'
insmod: ERROR: could not insert module atlantic.ko: Unknown symbol in module
dpkg: error processing package atlantic (--configure):
 installed atlantic package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 atlantic
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

CMDRZOD, 2020-01-04:
I'm seeing this as well. Exact same issue on Ubuntu 18.04.03 server. Looks like I'm using the same kernel as well.

cail, 2020-01-06:
This is most probably because .deb installation sequence does not take into account the module dependencies.
As a workaround please try doing `modprobe ptp crc_itu_t` and then repeat installation.

But please be aware that such installation will not always override inbox driver. Inbox driver usage is a preferred way for majority of regular users.

themightyoarfish, 2020-01-06:
By inbox driver you mean a driver bundled with Ubuntu? I don't seem to have one, at least the interface does not show up in ifconfig (lspci Shows the adapter).

CMDRZOD, 2020-01-10:
I just tried the modprobe command listed above. Unfortunately the same error still appears after bulding the module from source.

jason-edstrom, 2020-06-17:
I'm getting this error when trying to install the driver on Debian (specifically latest version of Open Media Vault).

jason-edstrom, 2020-06-17:
```Setting up atlantic (1592352125) ...
make -j4 -C /lib/modules/5.5.0-0.bpo.2-amd64/build M=/var/build_aq_drv/1592352125/Linux modules
make[1]: Entering directory '/usr/src/linux-headers-5.5.0-0.bpo.2-amd64'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.5.0-0.bpo.2-amd64'
insmod: ERROR: could not insert module atlantic.ko: Unknown symbol in module
dpkg: error processing package atlantic (--configure):
 installed atlantic package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 atlantic
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

cail, 2020-06-17:
Can you paste here your dmesg log? it should have error message in it.

vasilevnikitad, 2021-04-04:
When I turned off secure boot in BIOS, I was able to do 
```sh
sudo apt-get install ./atlantic_1578154949.deb 
```

tslater, 2023-09-29:
Isn't secure boot for windows?