Aquantia/AQtion #14: Wake-on-magic-packet with zeroed out SecureOn password?

charlessuh, 2020-02-18:
Hi.

I'm wondering if the driver is not properly configuring the firmware to wake on a magic packet, if an optional zeroed out SecureOn password is present at the end of the packet (`00:00:00:00:00`)? 

It appears I can wake a machine via `etherwake 00:17:b6:00:aa:bb` but not `etherwake -p 00:00:00:00:00:00 00:17:b6:00:aa:bb`. (Ignore that my MAC is not valid as I've replaced the last 2 bytes with `aa:bb`.)

The reason I'm running into this use case is that I'd like to rely on an Apple device to send a magic packet, and it appears it always send out zeroed out 6-byte password?

https://github.com/Apple-FOSS-Mirror/mDNSResponder/blob/b1b926e85440b53182d26715ece2fb67eac24f9e/mDNSMacOSX/helper.c#L2368

cail, 2020-02-18:
Hi, from our knowledge password in magic packet is not preventing AQC device from resume.
We however validated this using linux etherwake with -p option.

Could you capture a magic packet dump from apple device to see if its well formed?

charlessuh, 2020-02-18:
Thanks for validating that it works on your end.

I'll try to see if there might be a root cause elsewhere.

But FYI I have a AQN-100 identified as `Aquantia Corp. Device 00b1 (rev 02)` with:

```
driver: atlantic
version: 5.5.3-050503-generic-kern
firmware-version: 3.1.100
```

and here's two [pcaps](https://drive.google.com/file/d/1pXiDfnoCF2eQ6Bfx5F43ezzUZq5pWODv/view?usp=sharing) of one packet which works and another which doesn't work for me (both packets were generated using `etherwake`.)

cail, 2020-02-19:
Can you try installing the out of tree driver from this repository?

charlessuh, 2020-02-19:
Yep, I'm seeing the same behavior compiling and installing the driver from this repository:

```
driver: atlantic
version: 2.3.4.0
firmware-version: 3.1.100
```

I've also tried direct-connecting two computers together with static IPs (to eliminate any router or switches that were in the path) and seeing the same behavior.

Just to clarify, you are seeing successful wakes sending a 6-byte password, e.g:

```
etherwake -i eth1.10 -p 00:00:00:00:00:00 00:17:b6:00:aa:bb
 The Magic packet password is 00 00 00 00 (6).
```

where the target machine has wol set to only `g`:

```
Supports Wake-on: pg
Wake-on: g
```

If so, I'm wondering if perhaps the firmware version I'm running might also have an impact -- am I running the latest one?

Here's a [pcap](https://drive.google.com/open?id=1VdPWPw34RQvvVXL3FzvtaJjVgat1bJ6J) of  mDNSResponder on an Apple device sending a WOL packet, which sets a 6-byte zeroed out password and why I'm trying to get this use case to work.

cail, 2020-02-26:
> Just to clarify, you are seeing successful wakes sending a 6-byte password, e.g:

Yes. But that could be abit different setup of FW. I'll try to push this to investigate internally.