Aquantia/AQtion #18: Kernel crash on Debian 10

vvv-ca, 2020-08-28:
Hi,

I just tried the 2.4.3.0_207 driver from the marvel website on my laptop and the module crashes in insmod. Running Debian 10 with the QNAP QNA-T310G1S (AQC107 chip).

```
[691436.778002] ------------[ cut here ]------------
[691436.778005] WARNING: CPU: 6 PID: 239155 at net/ethtool/common.c:348 ethtool_check_ops+0x14/0x20
[691436.778005] Modules linked in: atlantic(OE+) ptp pps_core macsec crc_itu_t exfat sd_mod sg uas usb_storage cpuid ctr ccm tun xt_MASQUERADE nf_conntrack_netlink xfrm_user xfrm_algo nft_chain_nat nf_nat br_netfilter bridge stp llc vmnet(OE) fuse vboxnetadp(OE) vmw_vsock_vmci_transport vsock vboxnetflt(OE) vmw_vmci vmmon(OE) vboxdrv(OE) typec_displayport ipmi_devintf ipmi_msghandler overlay bnep wacom hid_multitouch nf_log_ipv6 ip6t_REJECT nf_reject_ipv6 ip6_tables xt_state nf_log_ipv4 nf_log_common nft_limit dell_rbtn nft_counter ipt_REJECT nf_reject_ipv4 x86_pkg_temp_thermal intel_powerclamp snd_sof_pci xt_multiport snd_hda_codec_hdmi snd_sof_intel_hda_common snd_sof_intel_hda xt_tcpudp snd_sof_intel_byt snd_sof_intel_ipc xt_LOG snd_sof xt_limit snd_sof_xtensa_dsp rtsx_pci_sdmmc xt_recent snd_soc_skl xt_addrtype mmc_core snd_soc_hdac_hda xt_conntrack kvm_intel nf_conntrack iwlmvm snd_hda_ext_core nf_defrag_ipv6 snd_soc_sst_ipc nf_defrag_ipv4 iTCO_wdt snd_soc_sst_dsp nft_compat
[691436.778023]  iTCO_vendor_support mei_wdt dell_laptop snd_soc_acpi_intel_match dell_wmi nf_tables watchdog snd_hda_codec_realtek i2c_designware_platform snd_soc_acpi i2c_designware_core kvm dell_smbios nfnetlink mac80211 snd_hda_codec_generic intel_rapl_msr irqbypass intel_wmi_thunderbolt wmi_bmof mxm_wmi nls_ascii dell_wmi_descriptor crc32_pclmul snd_soc_core ledtrig_audio libarc4 dcdbas nls_cp437 dell_smm_hwmon snd_compress ghash_clmulni_intel rapl vfat efi_pstore fat intel_cstate iwlwifi snd_hda_intel intel_uncore snd_intel_dspcfg snd_hda_codec efivars pcspkr snd_usb_audio cfg80211 rtsx_pci snd_hda_core snd_usbmidi_lib i2c_i801 snd_hwdep snd_rawmidi snd_seq_device cdc_ether snd_pcm usbnet r8152 snd_timer snd soundcore mii tpm_crb thunderbolt ahci libahci libata scsi_mod cdc_acm btusb uvcvideo btrtl btbcm btintel videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 bluetooth videobuf2_common videodev mc drbg ansi_cprng ecdh_generic ecc rfkill intel_lpss_pci mei_me intel_lpss idma64 mei
[691436.778041]  mfd_core processor_thermal_device intel_rapl_common intel_soc_dts_iosf intel_pch_thermal ucsi_acpi typec_ucsi typec i2c_hid button battery tpm_tis tpm_tis_core int3403_thermal int340x_thermal_zone tpm int3400_thermal wmi dell_smo8800 intel_hid acpi_thermal_rel ac acpi_pad rng_core sparse_keymap bbswitch(OE) coretemp nfsd ecryptfs parport_pc auth_rpcgss ppdev nfs_acl lockd lp grace parport sunrpc efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 hid_logitech_hidpp dm_crypt dm_mod raid10 raid456 libcrc32c crc32c_generic async_raid6_recov async_memcpy async_pq async_xor xor async_tx hid_logitech_dj raid6_pq raid1 raid0 multipath linear md_mod hid_generic usbhid hid joydev i915 crc32c_intel i2c_algo_bit drm_kms_helper xhci_pci xhci_hcd cec nvme aesni_intel drm usbcore psmouse glue_helper nvme_core libaes crypto_simd cryptd evdev t10_pi serio_raw crc_t10dif crct10dif_generic crct10dif_pclmul crct10dif_common usb_common video [last unloaded: atlantic]
[691436.778059] CPU: 6 PID: 239155 Comm: insmod Tainted: G     U  W  OE     5.7.0-2-amd64 #1 Debian 5.7.10-1
[691436.778059] Hardware name: Dell Inc. Precision 5540/0FMYX6, BIOS 1.8.1 07/03/2020
[691436.778060] RIP: 0010:ethtool_check_ops+0x14/0x20
[691436.778061] Code: bc f4 ff ff ff eb e6 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 0f 1f 44 00 00 31 c0 48 83 7f 70 00 74 0d 8b 17 85 d2 75 07 <0f> 0b b8 ea ff ff ff c3 0f 1f 40 00 0f 1f 44 00 00 4c 8b 8f f8 01
[691436.778061] RSP: 0018:ffffb0ec05f4ba60 EFLAGS: 00010246
[691436.778062] RAX: 0000000000000000 RBX: ffff93a5fe15a6c0 RCX: 0000000000000040
[691436.778063] RDX: 0000000000000000 RSI: 00000000fffffe01 RDI: ffffffffc1893c80
[691436.778063] RBP: ffff93a5e7b47000 R08: ffff93a6193fa1c8 R09: ffff93a6193fa1c8
[691436.778063] R10: 0000000000000dc0 R11: 0000000000000000 R12: ffff93a5fe15a740
[691436.778064] R13: ffffffffb3ef3c80 R14: 0000000000000020 R15: ffff93a5e7b47000
[691436.778064] FS:  00007f6c4f2fd540(0000) GS:ffff93a61c580000(0000) knlGS:0000000000000000
[691436.778065] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[691436.778065] CR2: 000017d6080bb000 CR3: 000000068aafc001 CR4: 00000000003606e0
[691436.778065] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[691436.778066] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[691436.778066] Call Trace:
[691436.778069]  register_netdevice+0x6a/0x580
[691436.778071]  register_netdev+0x1c/0x40
[691436.778078]  aq_nic_ndev_register+0x214/0x230 [atlantic]
[691436.778083]  aq_pci_probe+0x35b/0x3f0 [atlantic]
[691436.778085]  local_pci_probe+0x42/0x80
[691436.778086]  pci_device_probe+0x104/0x1a0
[691436.778088]  really_probe+0x147/0x3c0
[691436.778089]  driver_probe_device+0xb6/0x100
[691436.778090]  device_driver_attach+0x53/0x60
[691436.778091]  __driver_attach+0x8a/0x150
[691436.778091]  ? device_driver_attach+0x60/0x60
[691436.778092]  ? device_driver_attach+0x60/0x60
[691436.778093]  bus_for_each_dev+0x78/0xc0
[691436.778094]  bus_add_driver+0x14d/0x1f0
[691436.778095]  driver_register+0x6c/0xc0
[691436.778096]  ? 0xffffffffc0a47000
[691436.778100]  aq_ndev_init_module+0x49/0x1000 [atlantic]
[691436.778102]  do_one_initcall+0x46/0x200
[691436.778104]  ? _cond_resched+0x15/0x30
[691436.778105]  ? kmem_cache_alloc_trace+0x161/0x220
[691436.778106]  ? do_init_module+0x23/0x230
[691436.778107]  do_init_module+0x5c/0x230
[691436.778108]  load_module+0x234c/0x24f0
[691436.778110]  ? __do_sys_finit_module+0xaa/0x110
[691436.778110]  __do_sys_finit_module+0xaa/0x110
[691436.778112]  do_syscall_64+0x52/0x180
[691436.778113]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[691436.778114] RIP: 0033:0x7f6c4f41ea79
[691436.778115] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d e7 53 0c 00 f7 d8 64 89 01 48
[691436.778115] RSP: 002b:00007ffdaf29d4a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[691436.778116] RAX: ffffffffffffffda RBX: 0000562c136a2780 RCX: 00007f6c4f41ea79
[691436.778116] RDX: 0000000000000000 RSI: 0000562c1315a358 RDI: 0000000000000003
[691436.778116] RBP: 0000000000000000 R08: 0000000000000000 R09: 00007f6c4f4e8640
[691436.778117] R10: 0000000000000003 R11: 0000000000000246 R12: 0000562c1315a358
[691436.778117] R13: 0000000000000000 R14: 0000562c136a2760 R15: 0000000000000000
[691436.778118] ---[ end trace 43f1e3ec81833543 ]---
```
The adapter works just fine with the mainline Linux driver but that one lacks PTP support.
Any help is greatly appreciated!

Thanks!

cail, 2020-08-28:
Thanks for the reporting, thats incompat with newer kernels.

Please make this patch:

const struct ethtool_ops aq_ethtool_ops = {
+	.supported_coalesce_params = ETHTOOL_COALESCE_USECS |
+				     ETHTOOL_COALESCE_MAX_FRAMES,
	.get_link            = aq_ethtool_get_link,

we'll update github lateron

vvv-ca, 2020-08-28:
Excellent, thanks!

closed 2020-08-28 by vvv-ca