Aquantia/AQtion #19: PTP support on AQC107

vvv-ca, 2020-08-28:
Hi,

I'm trying to get PTP working on a QNAP QNA-T310G1S adapter which uses the AQC107 chip but ethtool returns no support whatsoever:
```
ethtool -T ens4
Time stamping parameters for ens4:
Capabilities:
        software-transmit
        software-receive
        software-system-clock
PTP Hardware Clock: none
Hardware Transmit Timestamp Modes: none
Hardware Receive Filter Modes: none
```
This is with module build from master here and 5.7.10-1 kernel. Chip is on firmware 3.1.109, ethtool is version 5.8

```
driver: atlantic
version: 5.7.0-2-amd64
firmware-version: 3.1.109
expansion-rom-version: 
bus-info: 0000:0a:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: yes
```

Is there anything I can do to troubleshoot this?
Thanks!

cail, 2020-08-28:
Plz either use latest kernel, or out of box driver from github.

closed 2020-08-28 by cail

vvv-ca, 2020-08-28:
But I did use the out of box driver from github:

version: 5.7.0-2-amd642.4.7.0

cail, 2020-08-28:
No, this version indicates thats a inbox. OOB should have 2.4.3

vvv-ca, 2020-08-28:
I built it from this github repo code which is 2.4.7. 2.4.3 is the code on the Marvell site. For whatever reason, the ethtool -i command pre-pends the kernel version to the driver version and reports the driver version as 5.7.0-2-amd642.4.7.0. Here is the modinfo as well:
```
filename:       /storage/code/AQtion/atlantic.ko
description:    Marvell (Aquantia) Corporation(R) Network Driver
author:         Marvell
version:        2.4.7.0
license:        GPL v2
firmware:       mrvl/91B1.fw
firmware:       mrvl/87B1.fw
firmware:       mrvl/80B1.fw
srcversion:     C4A5212D008FEA13D82CF1E
alias:          pci:v00001D6Ad000093C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000094C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000011C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000034C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000014C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000004C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000092B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000091B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000089B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000088B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000087B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000080B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000011B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000009B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000008B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000007B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D109sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D108sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D107sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D100sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad00000001sv*sd*bc*sc*i*
depends:        crc-itu-t,macsec,ptp
retpoline:      Y
name:           atlantic
vermagic:       5.7.0-2-amd64 SMP mod_unload modversions 
parm:           aq_ptp_offset_forced:Force to use the driver parameters (uint)
parm:           aq_ptp_offset_100:PTP offset for 100M (uint)
parm:           aq_ptp_offset_1000:PTP offset for 1G (uint)
parm:           aq_ptp_offset_2500:PTP offset for 2,5G (uint)
parm:           aq_ptp_offset_5000:PTP offset for 5G (uint)
parm:           aq_ptp_offset_10000:PTP offset for 10G (uint)
parm:           aq_ptp_gpio_hightime:PTP GPIO high time (uint)
parm:           sleep_delay:uint
parm:           aq_itr:Interrupt throttling mode (uint)
parm:           aq_itr_tx:TX interrupt throttle rate (uint)
parm:           aq_itr_rx:RX interrupt throttle rate (uint)
parm:           aq_rxpageorder:RX page order override (uint)
parm:           aq_rx_refill_thres:RX refill threshold (uint)
parm:           aq_fw_did:Use FW image for this DID (array of uint)
parm:           aq_fw_sid:Use provisioning data for this SID (array of uint)
parm:           aq_force_host_boot:Force host boot (array of uint)
parm:           aq_enable_wa:Quirk bits to enable HW workarounds (int)
parm:           aq_enable_ptp:Enable PTP (bool)
```

vvv-ca, 2020-08-28:
For future reference, the QNA-T310G1S uses the AQC100S which is a Ethernet MAC with no PTP support. The AQC100S seems to connect directly to the SFP modules via XFI so no PHY either. Thus no PTP support.
It does seem like the driver should provide at least software PTP timestamps.