Aquantia/AQtion #24: Asus XG-C100F won't detect SFP-10G-SR transceiver

mwpastore, 2020-12-23:
I've compiled and installed this module and unfortunately an Asus XG-C100F card won't detect a SFP-10G-SR ([FS.COM "Generic"](https://www.fs.com/products/74668.html)) transceiver. It's my understanding that these should be compatible. Any suggestions?

```
# ethtool enp6s0
Settings for enp6s0:
	Supported ports: [ FIBRE ]
	Supported link modes:   100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: Unknown!
	Duplex: Unknown! (255)
	Auto-negotiation: on
	Port: FIBRE
	PHYAD: 0
	Transceiver: internal
	Supports Wake-on: pg
	Wake-on: g
        Current message level: 0x00000005 (5)
                               drv link
	Link detected: no
# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.10
Release:	20.10
Codename:	groovy
# uname -a
Linux hera 5.8.0-33-generic #36-Ubuntu SMP Wed Dec 9 09:14:40 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
# modinfo atlantic | head
filename:       /lib/modules/5.8.0-33-generic/updates/dkms/atlantic.ko
description:    Marvell (Aquantia) Corporation(R) Network Driver
author:         Marvell
version:        2.4.10.0
license:        GPL v2
firmware:       mrvl/91B1.fw
firmware:       mrvl/87B1.fw
firmware:       mrvl/80B1.fw
srcversion:     645AFFA53090DE964B4BBED
alias:          pci:v00001D6Ad000093C0sv*sd*bc*sc*i*
```

Thank you in advance!

cail, 2020-12-24:
Hi,

Normally all the standard SFP+ modules should be OK, but I know there exist some incompats or even black lists.
Unfortunately can't do nothing here from driver side (

mwpastore, 2020-12-27:
Thanks @cail, I appreciate the quick response. Who can I reach out to for hardware support? Asus?

cail, 2020-12-27:
You may try Asus, sure. But the chip design is under Marvell...

petrgru, 2021-02-09:
same problem try 2 cooper cable a same result
   Supported FEC modes: Not reported
        Advertised link modes:  100baseT/Full 
                                1000baseT/Full 
                                10000baseT/Full 
                                2500baseT/Full 
                                5000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Port: FIBRE
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pg
        Wake-on: g
        Current message level: 0x00000005 (5)
                               drv link
        Link detected: no
root@superm2:~# ethtool -i enp2s0
driver: atlantic
version: 2.4.10.0
firmware-version: 3.1.109
expansion-rom-version: 
bus-info: 0000:02:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: yes

microtik cloud switch , cisco ... ( switch to switch working)

antoniopll, 2022-12-13:
Another one here with this wire:
[10 GB/s SFP+ DAC Cable 1-metro Pasivo, Compatible con Cisco SFP-H10GB-CU1M, Ubiquiti, Netgear, D-Link, TP-Link, Zyxel,and Other Open Switches] from H!Fiber.com