Aquantia/AQtion #26: Solo10G SFP+ (Thunderbolt 3 Edition): A lot of dmesg stacktrace errors

vasilevnikitad, 2021-04-04:
I got Solo10G SFP+ (Thunderbolt 3 Edition) and it seems like working properly but my dmesg is full off stacktrace errors when i connect it.

```sh
$ lsb_release -a 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.10
Release:	20.10
Codename:	groovy

$ uname -a
Linux xps 5.8.0-48-generic #54-Ubuntu SMP Fri Mar 19 14:25:20 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
$ sudo modinfo atlantic 
filename:       /lib/modules/5.8.0-48-generic/updates/dkms/atlantic.ko
description:    Marvell (Aquantia) Corporation(R) Network Driver
author:         Marvell
version:        2.4.10.0
license:        GPL v2
firmware:       mrvl/91B1.fw
firmware:       mrvl/87B1.fw
firmware:       mrvl/80B1.fw
srcversion:     85990593A7F718CBEAF2DB2
alias:          pci:v00001D6Ad000093C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000094C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000011C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000034C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000014C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000004C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000092B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000091B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000089B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000088B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000087B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000080B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000011B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000009B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000008B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000007B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D109sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D108sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D107sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D100sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad00000001sv*sd*bc*sc*i*
depends:        crc-itu-t,macsec
retpoline:      Y
name:           atlantic
vermagic:       5.8.0-48-generic SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         ubuntu Secure Boot Module Signature key
sig_key:        0C:40:9B:A6:38:F4:57:25:30:9D:86:0F:7B:CA:2F:F5:0D:74:DB:6C
sig_hashalgo:   sha512
signature:      87:E4:66:8E:4A:F1:33:8A:3C:0A:0D:08:CF:B4:8D:D5:F1:ED:21:97:
		5C:CE:63:B9:51:B7:96:20:89:31:41:4E:B0:85:5A:1D:E9:FE:44:98:
		D9:39:39:B3:BF:61:7E:5B:87:24:DC:8D:E9:86:A8:89:51:D7:8F:41:
		17:70:76:A3:41:71:D0:06:BF:DF:E2:65:B5:9F:C2:05:8B:AD:A0:19:
		16:40:D1:8D:E0:D2:3B:E1:FD:A2:56:39:80:81:C5:04:F2:54:81:BC:
		99:E7:EC:FC:4F:01:A2:F8:66:F6:B4:E2:23:06:AF:D9:EA:64:1B:2E:
		71:05:6B:89:06:25:F5:35:C2:BE:F7:A5:E2:47:8C:44:15:76:B2:17:
		36:95:3C:8A:24:BC:0B:D2:3E:CC:2A:A3:A6:8C:BE:AD:B5:42:A7:4A:
		9F:31:B0:C6:2E:7C:4A:65:73:CB:51:B0:7E:9F:EA:B0:83:91:66:9F:
		96:70:8E:64:12:A6:40:7B:68:84:B1:EE:50:D5:14:C7:7B:66:A6:4A:
		3E:8C:48:44:89:37:B0:D1:B4:8D:A2:A7:1A:EC:C9:E4:BE:AB:C8:78:
		49:DC:64:B0:70:E6:DF:C9:3A:42:34:5A:6A:D1:C2:21:57:81:97:16:
		D1:7B:F8:40:ED:42:07:FE:51:45:52:ED:76:48:4A:44
parm:           aq_ptp_offset_forced:Force to use the driver parameters (uint)
parm:           aq_ptp_offset_100:PTP offset for 100M (uint)
parm:           aq_ptp_offset_1000:PTP offset for 1G (uint)
parm:           aq_ptp_offset_2500:PTP offset for 2,5G (uint)
parm:           aq_ptp_offset_5000:PTP offset for 5G (uint)
parm:           aq_ptp_offset_10000:PTP offset for 10G (uint)
parm:           aq_ptp_gpio_hightime:PTP GPIO high time (uint)
parm:           sleep_delay:uint
parm:           aq_itr:Interrupt throttling mode (uint)
parm:           aq_itr_tx:TX interrupt throttle rate (uint)
parm:           aq_itr_rx:RX interrupt throttle rate (uint)
parm:           aq_rxpageorder:RX page order override (uint)
parm:           aq_rx_refill_thres:RX refill threshold (uint)
parm:           aq_fw_did:Use FW image for this DID (array of uint)
parm:           aq_fw_sid:Use provisioning data for this SID (array of uint)
parm:           aq_force_host_boot:Force host boot (array of uint)
parm:           aq_enable_wa:Quirk bits to enable HW workarounds (int)
parm:           aq_enable_ptp:Enable PTP (bool)

```

vasilevnikitad, 2021-04-04:
dmesg:
```sh
dmesg: 
```sh
[187673.190325] thunderbolt 0-1: new device found, vendor=0x8 device=0x36
[187673.190326] thunderbolt 0-1: Sonnet Technologies, Inc Solo 10G SFP+ Thunderbolt 3 Edition
[187673.440677] pcieport 0000:03:01.0: pciehp: Slot(1): Card present
[187673.440678] pcieport 0000:03:01.0: pciehp: Slot(1): Link Up
[187673.576497] pci 0000:05:00.0: [8086:15da] type 01 class 0x060400
[187673.576679] pci 0000:05:00.0: enabling Extended Tags
[187673.576966] pci 0000:05:00.0: supports D1 D2
[187673.576967] pci 0000:05:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[187673.577192] pci 0000:05:00.0: 8.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x4 link at 0000:03:01.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[187673.577268] pci 0000:05:00.0: Adding to iommu group 17
[187673.578925] pcieport 0000:03:01.0: ASPM: current common clock configuration is inconsistent, reconfiguring
[187673.579029] pci 0000:05:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[187673.579278] pci 0000:06:01.0: [8086:15da] type 01 class 0x060400
[187673.579489] pci 0000:06:01.0: enabling Extended Tags
[187673.579760] pci 0000:06:01.0: supports D1 D2
[187673.579760] pci 0000:06:01.0: PME# supported from D0 D1 D2 D3hot D3cold
[187673.580048] pci 0000:06:01.0: Adding to iommu group 17
[187673.581814] pci 0000:05:00.0: PCI bridge to [bus 06-39]
[187673.581832] pci 0000:05:00.0:   bridge window [io  0x0000-0x0fff]
[187673.581841] pci 0000:05:00.0:   bridge window [mem 0x00000000-0x000fffff]
[187673.581859] pci 0000:05:00.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[187673.581864] pci 0000:06:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[187673.582087] pci 0000:07:00.0: [1d6a:80b1] type 00 class 0x020000
[187673.582210] pci 0000:07:00.0: reg 0x10: [mem 0x00000000-0x0000ffff 64bit]
[187673.582252] pci 0000:07:00.0: reg 0x18: [mem 0x00000000-0x00000fff 64bit]
[187673.582294] pci 0000:07:00.0: reg 0x20: [mem 0x00000000-0x003fffff 64bit]
[187673.582317] pci 0000:07:00.0: reg 0x30: [mem 0x00000000-0x0003ffff pref]
[187673.582345] pci 0000:07:00.0: enabling Extended Tags
[187673.582598] pci 0000:07:00.0: supports D1 D2
[187673.582598] pci 0000:07:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[187673.582752] pci 0000:07:00.0: 8.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x4 link at 0000:03:01.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[187673.582803] pci 0000:07:00.0: Adding to iommu group 17
[187673.583814] pci 0000:06:01.0: PCI bridge to [bus 07-39]
[187673.583832] pci 0000:06:01.0:   bridge window [io  0x0000-0x0fff]
[187673.583840] pci 0000:06:01.0:   bridge window [mem 0x00000000-0x000fffff]
[187673.583858] pci 0000:06:01.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[187673.583859] pci_bus 0000:07: busn_res: [bus 07-39] end is updated to 07
[187673.583868] pci_bus 0000:06: busn_res: [bus 06-39] end is updated to 07
[187673.583882] pci 0000:05:00.0: BAR 14: assigned [mem 0xd4000000-0xe9efffff]
[187673.583883] pci 0000:05:00.0: BAR 15: assigned [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.583884] pci 0000:05:00.0: BAR 13: assigned [io  0x5000-0x5fff]
[187673.583885] pci 0000:06:01.0: BAR 14: assigned [mem 0xd4000000-0xe9efffff]
[187673.583886] pci 0000:06:01.0: BAR 15: assigned [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.583886] pci 0000:06:01.0: BAR 13: assigned [io  0x5000-0x5fff]
[187673.583888] pci 0000:07:00.0: BAR 4: assigned [mem 0xd4000000-0xd43fffff 64bit]
[187673.583924] pci 0000:07:00.0: BAR 6: assigned [mem 0xd4400000-0xd443ffff pref]
[187673.583925] pci 0000:07:00.0: BAR 0: assigned [mem 0xd4440000-0xd444ffff 64bit]
[187673.583958] pci 0000:07:00.0: BAR 2: assigned [mem 0xd4450000-0xd4450fff 64bit]
[187673.583991] pci 0000:06:01.0: PCI bridge to [bus 07]
[187673.583996] pci 0000:06:01.0:   bridge window [io  0x5000-0x5fff]
[187673.584009] pci 0000:06:01.0:   bridge window [mem 0xd4000000-0xe9efffff]
[187673.584017] pci 0000:06:01.0:   bridge window [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.584034] pci 0000:05:00.0: PCI bridge to [bus 06-07]
[187673.584038] pci 0000:05:00.0:   bridge window [io  0x5000-0x5fff]
[187673.584051] pci 0000:05:00.0:   bridge window [mem 0xd4000000-0xe9efffff]
[187673.584060] pci 0000:05:00.0:   bridge window [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.584076] pcieport 0000:03:01.0: PCI bridge to [bus 05-39]
[187673.584079] pcieport 0000:03:01.0:   bridge window [io  0x5000-0x5fff]
[187673.584086] pcieport 0000:03:01.0:   bridge window [mem 0xd4000000-0xe9efffff]
[187673.584091] pcieport 0000:03:01.0:   bridge window [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.584128] pcieport 0000:05:00.0: enabling device (0000 -> 0003)
[187673.584541] pcieport 0000:06:01.0: enabling device (0000 -> 0003)
[187673.585008] atlantic 0000:07:00.0: enabling device (0000 -> 0002)
[187674.009131] ------------[ cut here ]------------
[187674.009136] WARNING: CPU: 1 PID: 159 at net/core/dev.c:8719 __dev_xdp_query.part.0+0x6f/0x80
[187674.009136] Modules linked in: nfnetlink_queue nfnetlink_log usbhid rc_cec ccm atlantic(O) macsec crc_itu_t xt_conntrack xt_MASQUERADE nf_conntrack_netlink xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c nf_tables nfnetlink br_netfilter bridge stp llc rfcomm aufs cmac algif_hash overlay algif_skcipher af_alg bnep nvidia_uvm(O) nvidia_drm(PO) nvidia_modeset(PO) binfmt_misc snd [... cut ...]
[187674.011571] RSP: 002b:00007ffd1881a568 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[187674.011572] RAX: ffffffffffffffda RBX: 00005625965c1dc0 RCX: 00007ff2e9ddefca
[187674.011572] RDX: 0000000000000020 RSI: 00005625965b15e0 RDI: 000000000000000f
[187674.011573] RBP: 00005625965b11b0 R08: 00007ffd1881a5f0 R09: 0000000000000010
[187674.011573] R10: 0000000000000000 R11: 0000000000000246 R12: 000000000000f084
[187674.011574] R13: 00007ffd1881a6f8 R14: 00007ffd1881a700 R15: 0000000000000000
[187674.011574] ---[ end trace 8e14f5a588d21daa ]---
```
```

shmerl, 2022-11-21:
Did this get better in recent kernels (like 6.1-rc?). I was planning to get Solo10G SFP+, but such huge amount of errors in dmesg looks troubling.

vasilevnikitad, 2022-11-21:
Well, I don't use it anymore (I don't need it for the project) but have used it for 1 year and didn't have any problems with that except this one. 
(One more problem with this adapter is that not every Laptops USB Type-C slot has enough power to power this adapter. Sometimes it's solved by switching to another slot in the Laptop and sometimes it's solved by plugging in power cable into the laptop).

shmerl, 2022-11-21:
How much power does it draw (not counting the SFP+ transceiver own power consumption)? I tried to find that info on their site and manuals, but it doesn't say anything.

vasilevnikitad, 2022-11-21:
I cannot say. It didn't bother me since it was used only for debugging purpose. And, unfortunately, I don't have it anymore :/

timblaktu, 2023-09-21:
@shmerl I'm using the Solo 10G SFP+ with a Lenovo P1 Gen5 intel and although it works fine at boot, eventually it stops working and produces kernel errors and panics like this at shutdown. I'm troubleshooting this now and looking for info. FWIW I'm on kernel 6.5.3-arch1-1 and my [linux-hardware probe results are here](https://linux-hardware.org/?probe=8c65f7a68a).

shmerl, 2023-11-12:
That probably disables it completely? May be it can be disabled just for the adapter individually?

shmerl, 2024-02-04:
If anyone needs it, I got the adapter (it shows in lspci like this:

```
47:00.0 Ethernet controller: Aquantia Corp. Device 80b1 (rev 02)
```

And it works fine with Linux 6.7.3. I got 10 Gbps connection without any errors in dmesg. I haven't used it extensively though so far.

shmerl, 2024-02-04:
Somewhat unrelated, but in the above `` I see this detail: `sudo modinfo atlantic`

```
firmware:       mrvl/91B1.fw
firmware:       mrvl/87B1.fw
firmware:       mrvl/80B1.fw
```

Does one need these firmware files and where do they come from? I can't find them in the Linux firmware repo:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/