Aquantia/AQtion #29: cannot compile

filippocastelli, 2021-05-25:
I'm trying to compile the drivers under Kubuntu 20.04 with kernel 5.11.0-17-generic.

`sudo make` fails with the following error trail.
The driver compiles just fine on another Kubuntu 2.04 install with kernel 5.8
Hope I'm not missing anything trivial

```
In file included from /home/phil/aquantia/AQtion-master/aq_ptp.c:20:
/home/phil/aquantia/AQtion-master/aq_nic.h:210:51: warning: ‘struct ethtool_regs’ declared inside parameter list will not be visible outside of this definition or declaration
  210 | int aq_nic_get_regs(struct aq_nic_s *self, struct ethtool_regs *regs, void *p);
      |                                                   ^~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_nic.h:227:18: warning: ‘struct ethtool_link_ksettings’ declared inside parameter list will not be visible outside of this definition or declaration
  227 |           struct ethtool_link_ksettings *cmd);
      |                  ^~~~~~~~~~~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_nic.h:229:23: warning: ‘struct ethtool_link_ksettings’ declared inside parameter list will not be visible outside of this definition or declaration
  229 |          const struct ethtool_link_ksettings *cmd);
      |                       ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/phil/aquantia/AQtion-master/aq_ptp.c:25:
/home/phil/aquantia/AQtion-master/aq_filters.h:25:30: error: field ‘aq_fsp’ has incomplete type
   25 |  struct ethtool_rx_flow_spec aq_fsp;
      |                              ^~~~~~
/home/phil/aquantia/AQtion-master/aq_filters.h:30:61: warning: ‘struct ethtool_rxnfc’ declared inside parameter list will not be visible outside of this definition or declaration
   30 | int aq_add_rxnfc_rule(struct aq_nic_s *aq_nic, const struct ethtool_rxnfc *cmd);
      |                                                             ^~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_filters.h:31:61: warning: ‘struct ethtool_rxnfc’ declared inside parameter list will not be visible outside of this definition or declaration
   31 | int aq_del_rxnfc_rule(struct aq_nic_s *aq_nic, const struct ethtool_rxnfc *cmd);
      |                                                             ^~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_filters.h:32:55: warning: ‘struct ethtool_rxnfc’ declared inside parameter list will not be visible outside of this definition or declaration
   32 | int aq_get_rxnfc_rule(struct aq_nic_s *aq_nic, struct ethtool_rxnfc *cmd);
      |                                                       ^~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_filters.h:33:60: warning: ‘struct ethtool_rxnfc’ declared inside parameter list will not be visible outside of this definition or declaration
   33 | int aq_get_rxnfc_all_rules(struct aq_nic_s *aq_nic, struct ethtool_rxnfc *cmd,
      |                                                            ^~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_ptp.c: In function ‘aq_ptp_dpath_enable’:
/home/phil/aquantia/AQtion-master/aq_ptp.c:1065:9: error: variable ‘cmd’ has initializer but incomplete type
 1065 |  struct ethtool_rxnfc cmd = { 0 };
      |         ^~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_ptp.c:1065:31: warning: excess elements in struct initializer
 1065 |  struct ethtool_rxnfc cmd = { 0 };
      |                               ^
/home/phil/aquantia/AQtion-master/aq_ptp.c:1065:31: note: (near initialization for ‘cmd’)
/home/phil/aquantia/AQtion-master/aq_ptp.c:1065:23: error: storage size of ‘cmd’ isn’t known
 1065 |  struct ethtool_rxnfc cmd = { 0 };
      |                       ^~~
/home/phil/aquantia/AQtion-master/aq_ptp.c:1079:8: error: invalid use of undefined type ‘struct ethtool_rx_flow_spec’
 1079 |     fsp->ring_cookie = rx_queue;
      |        ^~
/home/phil/aquantia/AQtion-master/aq_ptp.c:1080:8: error: invalid use of undefined type ‘struct ethtool_rx_flow_spec’
 1080 |     fsp->flow_type = UDP_V4_FLOW;
      |        ^~
In file included from /home/phil/aquantia/AQtion-master/aq_main.h:14,
                 from /home/phil/aquantia/AQtion-master/aq_main.c:10:
/home/phil/aquantia/AQtion-master/aq_nic.h:210:51: warning: ‘struct ethtool_regs’ declared inside parameter list will not be visible outside of this definition or declaration
  210 | int aq_nic_get_regs(struct aq_nic_s *self, struct ethtool_regs *regs, void *p);
      |                                                   ^~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_nic.h:227:18: warning: ‘struct ethtool_link_ksettings’ declared inside parameter list will not be visible outside of this definition or declaration
  227 |           struct ethtool_link_ksettings *cmd);
      |                  ^~~~~~~~~~~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_nic.h:229:23: warning: ‘struct ethtool_link_ksettings’ declared inside parameter list will not be visible outside of this definition or declaration
  229 |          const struct ethtool_link_ksettings *cmd);
      |                       ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/phil/aquantia/AQtion-master/aq_main.c:19:
/home/phil/aquantia/AQtion-master/aq_filters.h:25:30: error: field ‘aq_fsp’ has incomplete type
   25 |  struct ethtool_rx_flow_spec aq_fsp;
      |                              ^~~~~~
/home/phil/aquantia/AQtion-master/aq_filters.h:30:61: warning: ‘struct ethtool_rxnfc’ declared inside parameter list will not be visible outside of this definition or declaration
   30 | int aq_add_rxnfc_rule(struct aq_nic_s *aq_nic, const struct ethtool_rxnfc *cmd);
      |                                                             ^~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_filters.h:31:61: warning: ‘struct ethtool_rxnfc’ declared inside parameter list will not be visible outside of this definition or declaration
   31 | int aq_del_rxnfc_rule(struct aq_nic_s *aq_nic, const struct ethtool_rxnfc *cmd);
[... 411 lines cut ...]
      |         ^~
/home/phil/aquantia/AQtion-master/aq_nic.c:1275:27: error: ‘AUTONEG_ENABLE’ undeclared (first use in this function)
 1275 |  if (cmd->base.autoneg == AUTONEG_ENABLE) {
      |                           ^~~~~~~~~~~~~~
/home/phil/aquantia/AQtion-master/aq_nic.c:1280:8: error: ‘SPEED_10’ undeclared (first use in this function); did you mean ‘SPEED_5000’?
 1280 |   case SPEED_10:
      |        ^~~~~~~~
      |        SPEED_5000
/home/phil/aquantia/AQtion-master/aq_nic.c:1284:8: error: ‘SPEED_100’ undeclared (first use in this function); did you mean ‘SPEED_5000’?
 1284 |   case SPEED_100:
      |        ^~~~~~~~~
      |        SPEED_5000
/home/phil/aquantia/AQtion-master/aq_nic.c:1293:8: error: ‘SPEED_2500’ undeclared (first use in this function); did you mean ‘SPEED_5000’?
 1293 |   case SPEED_2500:
      |        ^~~~~~~~~~
      |        SPEED_5000
/home/phil/aquantia/AQtion-master/aq_nic.c:1301:8: error: ‘SPEED_10000’ undeclared (first use in this function); did you mean ‘SPEED_5000’?
 1301 |   case SPEED_10000:
      |        ^~~~~~~~~~~
      |        SPEED_5000
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:287: /home/phil/aquantia/AQtion-master/aq_nic.o] Errore 1
make[1]: *** [Makefile:1837: /home/phil/aquantia/AQtion-master] Errore 2
make: *** [Makefile:36: all] Errore 2
```

Forusim, 2021-06-07:
Have exact same issue on Kubuntu 21.04. 
Compiled on Linux Mint 20.01 (Ubuntu 20.04) without issues, but switched to kernel 5.11 for Intel Rocket Lake support.

cail, 2021-06-07:
Please try version 2.4.14 - should be fixed.

Forusim, 2021-06-07:
Thanks for quick reply, are the warnings expected?
```
user@host:~/Downloads/AQtion-master$ make
make -j4 -C /lib/modules/5.11.0-18-generic/build M="/home/user/Downloads/AQtion-master" modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-18-generic'
  CC [M]  /home/user/Downloads/AQtion-master/aq_main.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_nic.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_pci_func.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_vec.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_ring.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_ptp.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_filters.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_hw_utils.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_ethtool.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_drvinfo.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_trace.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_phy.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_compat.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_tsn.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_sysfs.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl/hw_atl_a0.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl/hw_atl_b0.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl/hw_atl_utils.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl/hw_atl_utils_fw2x.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl/hw_atl_llh.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl/hw_atl_fw_hostboot.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl/hw_atl_fw_image.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl2/hw_atl2_llh.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl2/hw_atl2.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl2/hw_atl2_utils.o
  CC [M]  /home/user/Downloads/AQtion-master/hw_atl2/hw_atl2_utils_fw.o
  CC [M]  /home/user/Downloads/AQtion-master/macsec/macsec_api.o
  CC [M]  /home/user/Downloads/AQtion-master/aq_macsec.o
  LD [M]  /home/user/Downloads/AQtion-master/atlantic.o
  MODPOST /home/user/Downloads/AQtion-master/Module.symvers
  CC [M]  /home/user/Downloads/AQtion-master/atlantic.mod.o
  LD [M]  /home/user/Downloads/AQtion-master/atlantic.ko
  BTF [M] /home/user/Downloads/AQtion-master/atlantic.ko
Skipping BTF generation for /home/user/Downloads/AQtion-master/atlantic.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-18-generic'
user@host:~/Downloads/AQtion-master$ sudo rmmod atlantic
user@host:~/Downloads/AQtion-master$ sudo make load
modprobe ptp
modprobe crc_itu_t
modprobe -q macsec || true
insmod ./atlantic.ko
user@host:~/Downloads/AQtion-master$ 
::1              ff00::0          ip6-allnodes     ip6-localnet     localhost        
host             ff02::1          ip6-allrouters   ip6-loopback     
fe00::0          ff02::2          ip6-localhost    ip6-mcastprefix  
user@host:~/Downloads/AQtion-master$ sudo make install
make[1]: Entering directory '/home/user/Downloads/AQtion-master'
atlantic.ko is in initramfs.
CAUTION! Updating initramfs is potentially dangerous.
Attempt initramfs update? [yN] y
update-initramfs: Generating /boot/initrd.img-5.11.0-18-generic
W: Possible missing firmware /lib/firmware/mrvl/91B1.fw for module atlantic
W: Possible missing firmware /lib/firmware/mrvl/87B1.fw for module atlantic
W: Possible missing firmware /lib/firmware/mrvl/80B1.fw for module atlantic
make[1]: Leaving directory '/home/user/Downloads/AQtion-master'
```

cail, 2021-06-07:
Yes, FW files are optional.

closed 2021-06-07 by cail