Aquantia/AQtion #33: Builds fail with `5.15` kernels

andrejpodzimek, 2021-11-19:
I'm using [this ArchLinux package](https://aur.archlinux.org/packages/atlantic-dkms). The DKMS build fails this way:

```
/var/lib/dkms/atlantic/2.4.14/build/aq_ethtool.c:1473:32: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ [-Werror=incompatible-pointer-types]
 1473 |         .get_coalesce        = aq_ethtool_get_coalesce,
      |                                ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/atlantic/2.4.14/build/aq_ethtool.c:1473:32: note: (near initialization for ‘aq_ethtool_ops.get_coalesce’)
/var/lib/dkms/atlantic/2.4.14/build/aq_ethtool.c:1474:32: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ [-Werror=incompatible-pointer-types]
 1474 |         .set_coalesce        = aq_ethtool_set_coalesce,
      |                                ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/atlantic/2.4.14/build/aq_ethtool.c:1474:32: note: (near initialization for ‘aq_ethtool_ops.set_coalesce’)
```

Is this a known issue? Is there an update?

robert-sc, 2022-01-27:
FWIW, you can probably simply add the missing parameters, as was done in the in-tree driver:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/net/ethernet/aquantia/atlantic/aq_ethtool.c?id=f3ccfda1931977b80267ba54070a1aeafa18f6ca

Also, note that 2.4.14 is not the latest Marvell/Aquantia release. You can find release 2.4.15 on the Marvell download size, by selecting e.g. "AQC107" in the "Part Number" selection dropdown:
https://www.marvell.com/support/downloads.html

cail, 2022-01-28:
pushed latest updates, should work with everything up to 5.17

closed 2022-01-28 by cail

Shekelme, 2022-05-30:
Isn't this a reason to make тусуыыфкн changes to `aq_ethtool.c` officially?

Shekelme, 2023-08-16:
```
Processing triggers for linux-image-5.15.0-79-generic (5.15.0-79.86~20.04.2) ...
/etc/kernel/postinst.d/dkms:
 * dkms: running auto installation service for kernel 5.15.0-79-generic

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j32 KERNELRELEASE=5.15.0-79-generic -C /lib/modules/5.15.0-79-generic/build M=/var/lib/dkms/atlantic/2.4.15.0/build...(bad exit status: 2)
ERROR (dkms apport): binary package for atlantic: 2.4.15.0 not found
Error! Bad return status for module build on kernel: 5.15.0-79-generic (x86_64)
Consult /var/lib/dkms/atlantic/2.4.15.0/build/make.log for more information.
   ...done.
```
make.log:
```
KMS make.log for atlantic-2.4.15.0 for kernel 5.15.0-79-generic (x86_64)
Wed 16 Aug 2023 09:46:47 PM MSK
make: Entering directory '/usr/src/linux-headers-5.15.0-79-generic'
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_main.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_nic.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_pci_func.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_vec.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_ring.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_ptp.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_filters.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_hw_utils.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_ethtool.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_drvinfo.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_trace.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_phy.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_compat.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_tsn.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_sysfs.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl/hw_atl_a0.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl/hw_atl_b0.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl/hw_atl_utils.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl/hw_atl_utils_fw2x.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl/hw_atl_llh.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl/hw_atl_fw_hostboot.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl/hw_atl_fw_image.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl2/hw_atl2_llh.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl2/hw_atl2.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl2/hw_atl2_utils.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/hw_atl2/hw_atl2_utils_fw.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/macsec/macsec_api.o
  CC [M]  /var/lib/dkms/atlantic/2.4.15.0/build/aq_macsec.o
/var/lib/dkms/atlantic/2.4.15.0/build/aq_ethtool.c:1473:25: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *>
 1473 |  .get_coalesce        = aq_ethtool_get_coalesce,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/atlantic/2.4.15.0/build/aq_ethtool.c:1473:25: note: (near initialization for ‘aq_ethtool_ops.get_coalesce’)
/var/lib/dkms/atlantic/2.4.15.0/build/aq_ethtool.c:1474:25: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *>
 1474 |  .set_coalesce        = aq_ethtool_set_coalesce,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/atlantic/2.4.15.0/build/aq_ethtool.c:1474:25: note: (near initialization for ‘aq_ethtool_ops.set_coalesce’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:297: /var/lib/dkms/atlantic/2.4.15.0/build/aq_ethtool.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1909: /var/lib/dkms/atlantic/2.4.15.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.0-79-generic'
```

texn0id, 2024-02-15:
I'm experiencing the same problem
`Linux version 5.15.0-94-generic (buildd@lcy02-amd64-118) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34)` 
Ubuntu 20.04

`sudo ./dkms.sh install`

> Creating symlink /var/lib/dkms/atlantic/2.4.14.0/source ->
>                  /usr/src/atlantic-2.4.14.0
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> make -j24 KERNELRELEASE=5.15.0-94-generic -C /lib/modules/5.15.0-94-generic/build M=/var/lib/dkms/atlantic/2.4.14.0/build....(bad exit status: 2)
> ERROR (dkms apport): binary package for atlantic: 2.4.14.0 not found
> Error! Bad return status for module build on kernel: 5.15.0-94-generic (x86_64)
> Consult /var/lib/dkms/atlantic/2.4.14.0/build/make.log for more information.

`make`

> make -j4 -C /lib/modules/5.15.0-94-generic/build M="/home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux" modules
> make[1]: Entering directory '/usr/src/linux-headers-5.15.0-94-generic'
>   CC [M]  /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux/aq_ethtool.o
> /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux/aq_ethtool.c:1473:25: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ [-Werror=incompatible-pointer-types]
>  1473 |  .get_coalesce        = aq_ethtool_get_coalesce,
>       |                         ^~~~~~~~~~~~~~~~~~~~~~~
> /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux/aq_ethtool.c:1473:25: note: (near initialization for ‘aq_ethtool_ops.get_coalesce’)
> /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux/aq_ethtool.c:1474:25: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ [-Werror=incompatible-pointer-types]
>  1474 |  .set_coalesce        = aq_ethtool_set_coalesce,
>       |                         ^~~~~~~~~~~~~~~~~~~~~~~
> /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux/aq_ethtool.c:1474:25: note: (near initialization for ‘aq_ethtool_ops.set_coalesce’)
> /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux/aq_ethtool.c:1486:1: fatal error: opening dependency file /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux/.aq_ethtool.o.d: Permission denied
>  1486 | };
>       | ^
> cc1: some warnings being treated as errors
> compilation terminated.
> make[2]: *** [scripts/Makefile.build:297: /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux/aq_ethtool.o] Error 1
> make[1]: *** [Makefile:1911: /home/aspod/LinuxDriver/AQC_Linux_Driver_2.4.14.0/atlantic/Linux] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-94-generic'
> make: *** [Makefile:36: all] Error 2