Aquantia/AQtion #34: Gigabit SFP module shows link up with correct rate, but doesn't receive any octets/packets

jeremyherbert, 2021-11-26:
Hi, 

I have a QNAP T310G1S (AQC100, PCIe over thunderbolt) running on an Intel NUC with Ubuntu 20.04. I am trying to use it with a gigabit (not 10G) SFP module. SFPs are Finisar FTLF8524P2BNV which I have tested as working with two Marvell 88E1111-based adapters as well as a Realtek-based switch (RTL9303).

As far as I can tell, the AQC100 is supposed to work with 1G adapters, as there are notes like this in the firmware changelog:
```
3.1.109
21 February 2020
-------------
Bug fix:
Fixed TX Datapath stuck caused by flow control
Enhancement:
[AQC-100]
Improved 1G optic link up time <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Changes:
LEDs interface now uses register 0x31C instead of 0x37C
```
Though I am not sure how to check the firmware version of the device.

Here is the module info:

```
filename:       /lib/modules/5.4.0-90-generic/kernel/drivers/net/ethernet/aquantia/atlantic/atlantic.ko
description:    aQuantia Corporation(R) Network Driver
author:         aQuantia
version:        5.4.0-90-generic-kern
license:        GPL v2
srcversion:     49C6E8981415B4E809ED675
alias:          pci:v00001D6Ad000092B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000091B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000089B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000088B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000087B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000080B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000011B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000009B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000008B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000007B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D109sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D108sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D107sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D100sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad00000001sv*sd*bc*sc*i*
depends:
retpoline:      Y
intree:         Y
name:           atlantic
vermagic:       5.4.0-90-generic SMP mod_unload modversions
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        39:76:0F:24:EA:8B:04:F4:F1:6A:A6:B7:5B:FC:76:15:B6:68:0D:E5
sig_hashalgo:   sha512
signature:      7F:63:10:30:84:64:D0:A7:EC:82:5D:12:CD:EA:87:4A:B4:93:13:26:
		5A:57:2A:83:5E:06:0A:BE:27:5D:4D:AD:E9:85:F0:70:E9:2D:94:EF:
		29:48:C6:B8:DE:36:96:87:48:20:34:72:21:52:2E:00:DC:FD:02:36:
		E8:2B:0C:FA:84:A7:C1:9B:58:6C:66:D3:FF:21:F1:C9:AE:23:A2:A2:
		1B:A8:C3:E7:1B:5A:C5:58:51:E4:0F:7D:55:BB:AF:7F:AD:52:2F:6F:
		49:2E:BB:4C:86:16:36:83:26:F3:09:7D:96:10:EF:DE:FE:27:3D:02:
		A4:43:F8:84:52:24:0E:CA:43:36:3F:7A:79:32:B8:17:FC:60:CB:AB:
		39:44:0B:4F:1A:D9:5A:BD:91:2B:E9:21:23:73:2E:D7:8D:C6:79:02:
		89:03:72:D1:96:EB:B1:22:64:EC:93:DF:16:5A:FC:B5:7F:28:3E:05:
		00:9C:97:3C:F6:7D:53:93:A6:1E:22:32:5E:98:E9:3C:1F:EC:37:F5:
		FB:56:CB:98:01:F7:03:24:DB:11:37:EA:16:A7:E9:3F:4B:1B:EF:0C:
		F8:98:CD:14:18:65:23:0B:50:01:DB:C6:F5:61:C8:6A:B0:D3:25:CA:
		97:FD:98:E6:A5:A9:71:55:C9:86:8B:57:6B:F7:A2:49:C5:C9:E2:46:
		74:E4:00:EE:A6:96:2C:90:C5:63:7D:B5:FA:E0:63:B8:0F:44:00:5D:
		25:90:14:3E:8C:1C:AD:1C:FB:B9:BB:06:4C:AA:72:0C:94:80:C0:4B:
		CF:C5:1D:7B:00:0B:C0:36:75:E7:A5:0B:51:2A:E8:22:4E:69:09:D4:
		6D:FF:27:ED:2E:9C:82:9D:FA:2D:2F:F7:39:F9:08:19:6D:6A:C8:BF:
		45:92:F4:72:B3:FE:B2:6F:1A:46:7C:40:BB:EF:8B:DE:D5:BF:18:76:
		A2:E8:1C:AD:0F:FC:41:50:6D:E9:FE:0D:75:F8:D3:5C:39:0E:2E:6C:
		08:AF:6A:7E:02:86:86:67:86:BF:E2:22:E8:6C:C8:67:30:DA:06:AC:
		B2:39:F1:D5:EC:6E:92:52:2C:F2:E8:D7:7B:A3:AD:75:FD:4F:86:AB:
		C5:B4:C3:73:36:50:A7:6E:FB:28:79:67:6A:74:94:86:17:73:82:13:
		C2:A1:93:08:08:49:7C:D4:61:08:EB:BD:AB:FE:05:7B:1E:82:F9:52:
		37:0A:EC:A4:BC:88:21:AC:40:5E:68:56:C7:6D:A9:16:81:53:90:79:
		3C:91:38:94:B6:BD:EB:6E:72:96:A9:1A:17:C5:D3:7B:C4:50:CF:76:
		29:25:00:81:3F:3B:B9:FF:F3:67:39:50
parm:           aq_itr:Interrupt throttling mode (uint)
parm:           aq_itr_tx:TX interrupt throttle rate (uint)
parm:           aq_itr_rx:RX interrupt throttle rate (uint)
```

The link rate is correctly detected by `ethtool`
```
Settings for ens1:
	Supported ports: [ FIBRE ]
	Supported link modes:   100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Supported pause frame use: Symmetric
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Advertised pause frame use: Symmetric
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: FIBRE
	PHYAD: 0
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: g
	Wake-on: d
	Link detected: yes
```

and if I unplug the fibre, then the link correctly goes down:
```
Settings for ens1:
	Supported ports: [ FIBRE ]
	Supported link modes:   100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Supported pause frame use: Symmetric
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Advertised pause frame use: Symmetric
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: Unknown!
	Duplex: Full
	Port: FIBRE
	PHYAD: 0
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: g
	Wake-on: d
	Link detected: no
```

I also get messages like this in `dmesg` when I unplug/replug:
```
[ 4495.900997] atlantic: link change old 1000 new 0
[ 4519.835802] atlantic: link change old 0 new 1000
```

The problem appears to be related to the fact that there are no inbound packets appearing (toward the AQC100) according to `ethtool -s`:
```
NIC statistics:
     InPackets: 0
     InUCast: 0
     InMCast: 0
     InBCast: 0
     InErrors: 0
     OutPackets: 29
     OutUCast: 0
     OutMCast: 23
     OutBCast: 6
     InUCastOctets: 0
     OutUCastOctets: 0
     InMCastOctets: 0
     OutMCastOctets: 3664
     InBCastOctets: 0
     OutBCastOctets: 384
     InOctets: 0
     OutOctets: 4048
     InPacketsDma: 0
     OutPacketsDma: 29
     InOctetsDma: 0
     OutOctetsDma: 3812
     InDroppedDma: 0
     Queue[0] InPackets: 0
     Queue[0] OutPackets: 0
     Queue[0] Restarts: 0
     Queue[0] InJumboPackets: 0
     Queue[0] InLroPackets: 0
     Queue[0] InErrors: 0
     Queue[1] InPackets: 0
     Queue[1] OutPackets: 3
     Queue[1] Restarts: 0
     Queue[1] InJumboPackets: 0
     Queue[1] InLroPackets: 0
     Queue[1] InErrors: 0
     Queue[2] InPackets: 0
     Queue[2] OutPackets: 15
     Queue[2] Restarts: 0
     Queue[2] InJumboPackets: 0
     Queue[2] InLroPackets: 0
     Queue[2] InErrors: 0
     Queue[3] InPackets: 0
     Queue[3] OutPackets: 0
     Queue[3] Restarts: 0
     Queue[3] InJumboPackets: 0
     Queue[3] InLroPackets: 0
     Queue[3] InErrors: 0
     Queue[4] InPackets: 0
     Queue[4] OutPackets: 3
     Queue[4] Restarts: 0
     Queue[4] InJumboPackets: 0
     Queue[4] InLroPackets: 0
     Queue[4] InErrors: 0
     Queue[5] InPackets: 0
     Queue[5] OutPackets: 0
     Queue[5] Restarts: 0
     Queue[5] InJumboPackets: 0
     Queue[5] InLroPackets: 0
     Queue[5] InErrors: 0
     Queue[6] InPackets: 0
     Queue[6] OutPackets: 8
     Queue[6] Restarts: 0
     Queue[6] InJumboPackets: 0
     Queue[6] InLroPackets: 0
     Queue[6] InErrors: 0
     Queue[7] InPackets: 0
     Queue[7] OutPackets: 0
     Queue[7] Restarts: 0
     Queue[7] InJumboPackets: 0
     Queue[7] InLroPackets: 0
     Queue[7] InErrors: 0
```

I assume that it knows the link is up because it can exchange ordered sets with the other end of the link and perform clock recovery. I have checked and the other end of the link (Marvell 88E1111) is indeed transmitting packets, but it also reports zero packets received.

Are there any known quirks with a configuration like this that I need to be aware of to get it to work?

cail, 2021-11-29:
Unfortunately that could be SFP dongle incompatibility. On a driver level - we don't have much to do to understand the root cause of the problem...

jeremyherbert, 2021-11-29:
Thanks for your reply; I suspected that might be the case. Is there some way I could contact the firmware support team about this? Do you know of any gigabit SFPs which are known to work (even if unsupported)?

cail, 2021-11-30:
You may try contacting official Marvell support, however I would say the chances to find any info are low. I don't know any of specific SFP models, may be this is also a thing support team can reveal.

jeremyherbert, 2021-11-30:
Thanks again. I will try my luck with Marvell support.

closed 2021-11-30 by jeremyherbert