Aquantia/AQtion #35: atlantic driver fails to load on Kernel 5.16.4 for XG-C100F SFP+

eitch, 2022-02-01:
Feb  1 23:13:13 mypc kernel: [    8.090504] RIP: 0010:aq_nic_start+0x363/0x370 [atlantic]
Feb  1 23:13:13 mypc kernel: [    8.090509] Code: cc 50 c0 e8 4f fa a1 d9 41 89 c4 85 c0 0f 88 57 fd ff ff 41 8b 8d 94 01 00 00 b8 01 00 00 00 d3 e0 41 09 85 f8 05 00 00 eb 84 <0f> 0b 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 55 48 89 e5
Feb  1 23:13:13 mypc kernel: [    8.090510] RSP: 0018:ffffad63c25bf468 EFLAGS: 00010202
Feb  1 23:13:13 mypc kernel: [    8.090512] RAX: 0000000000000008 RBX: 0000000000000008 RCX: ffff96f1a15b6000
Feb  1 23:13:13 mypc kernel: [    8.090513] RDX: 0000000000000010 RSI: ffffad63c18e5be8 RDI: ffff96f19fbbe428
Feb  1 23:13:13 mypc kernel: [    8.090513] RBP: ffffad63c25bf480 R08: 0000000020000800 R09: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.090514] R10: 0000000000000000 R11: 000000000000003f R12: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.090514] R13: ffff96f1a15b6980 R14: 0000000000000001 R15: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.090515] FS:  00007f8425114140(0000) GS:ffff97006ea00000(0000) knlGS:0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.090516] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb  1 23:13:13 mypc kernel: [    8.090517] CR2: 00007fa0f0df9710 CR3: 0000000109a5c000 CR4: 0000000000750ef0
Feb  1 23:13:13 mypc kernel: [    8.090517] PKRU: 55555554
Feb  1 23:13:13 mypc kernel: [    8.090518] Call Trace:
Feb  1 23:13:13 mypc kernel: [    8.090519]  <TASK>
Feb  1 23:13:13 mypc kernel: [    8.090521]  aq_ndev_open+0x49/0x70 [atlantic]
Feb  1 23:13:13 mypc kernel: [    8.090524]  __dev_open+0xef/0x180
Feb  1 23:13:13 mypc kernel: [    8.090527]  __dev_change_flags+0x188/0x1e0
Feb  1 23:13:13 mypc kernel: [    8.090528]  dev_change_flags+0x26/0x60
Feb  1 23:13:13 mypc kernel: [    8.090529]  do_setlink+0x908/0xc40
Feb  1 23:13:13 mypc kernel: [    8.090531]  ? __nla_validate_parse+0x4c/0x1a0
Feb  1 23:13:13 mypc kernel: [    8.090533]  __rtnl_newlink+0x597/0xa10
Feb  1 23:13:13 mypc kernel: [    8.090534]  ? __nla_put+0x20/0x30
Feb  1 23:13:13 mypc kernel: [    8.090535]  ? pskb_expand_head+0x75/0x310
Feb  1 23:13:13 mypc kernel: [    8.090537]  ? security_sock_rcv_skb+0x2f/0x50
Feb  1 23:13:13 mypc kernel: [    8.090539]  ? netlink_deliver_tap+0x49/0x240
Feb  1 23:13:13 mypc kernel: [    8.090541]  ? sk_filter_trim_cap+0xbf/0x220
Feb  1 23:13:13 mypc kernel: [    8.090542]  ? skb_queue_tail+0x48/0x50
Feb  1 23:13:13 mypc kernel: [    8.090544]  ? sock_def_readable+0x50/0x80
Feb  1 23:13:13 mypc kernel: [    8.090545]  ? __netlink_sendskb+0x42/0x50
Feb  1 23:13:13 mypc kernel: [    8.090547]  ? netlink_unicast+0x213/0x250
Feb  1 23:13:13 mypc kernel: [    8.090548]  ? rtnl_getlink+0x376/0x400
Feb  1 23:13:13 mypc kernel: [    8.090550]  ? security_capable+0x3d/0x60
Feb  1 23:13:13 mypc kernel: [    8.090552]  ? kmem_cache_alloc_trace+0x193/0x470
Feb  1 23:13:13 mypc kernel: [    8.090554]  rtnl_newlink+0x49/0x70
Feb  1 23:13:13 mypc kernel: [    8.090555]  rtnetlink_rcv_msg+0x160/0x3b0
Feb  1 23:13:13 mypc kernel: [    8.090557]  ? __mark_inode_dirty+0x105/0x350
Feb  1 23:13:13 mypc kernel: [    8.090559]  ? rtnl_calcit.isra.0+0x130/0x130
Feb  1 23:13:13 mypc kernel: [    8.090561]  netlink_rcv_skb+0x55/0x100
Feb  1 23:13:13 mypc kernel: [    8.090562]  rtnetlink_rcv+0x15/0x20
Feb  1 23:13:13 mypc kernel: [    8.090563]  netlink_unicast+0x1a8/0x250
Feb  1 23:13:13 mypc kernel: [    8.090565]  netlink_sendmsg+0x245/0x490
Feb  1 23:13:13 mypc kernel: [    8.090566]  ? _copy_from_user+0x2b/0x60
Feb  1 23:13:13 mypc kernel: [    8.090567]  sock_sendmsg+0x65/0x70
Feb  1 23:13:13 mypc kernel: [    8.090568]  ____sys_sendmsg+0x218/0x290
Feb  1 23:13:13 mypc kernel: [    8.090569]  ? copy_msghdr_from_user+0x5c/0x90
Feb  1 23:13:13 mypc kernel: [    8.090570]  ? addrconf_sysctl_forward+0x111/0x280
Feb  1 23:13:13 mypc kernel: [    8.090572]  ? _copy_from_iter+0x92/0x590
Feb  1 23:13:13 mypc kernel: [    8.090574]  ___sys_sendmsg+0x81/0xc0
Feb  1 23:13:13 mypc kernel: [    8.090575]  ? mntput_no_expire+0x53/0x260
Feb  1 23:13:13 mypc kernel: [    8.090576]  ? __call_rcu+0xac/0x270
Feb  1 23:13:13 mypc kernel: [    8.090578]  ? call_rcu+0xe/0x10
Feb  1 23:13:13 mypc kernel: [    8.090580]  ? __fget_light+0xc5/0x100
Feb  1 23:13:13 mypc kernel: [    8.090582]  __sys_sendmsg+0x62/0xb0
Feb  1 23:13:13 mypc kernel: [    8.090583]  __x64_sys_sendmsg+0x1f/0x30
Feb  1 23:13:13 mypc kernel: [    8.090584]  do_syscall_64+0x5c/0xc0
Feb  1 23:13:13 mypc kernel: [    8.090586]  ? do_syscall_64+0x69/0xc0
Feb  1 23:13:13 mypc kernel: [    8.090587]  ? asm_exc_page_fault+0x8/0x30
Feb  1 23:13:13 mypc kernel: [    8.090589]  entry_SYSCALL_64_after_hwframe+0x44/0xae
Feb  1 23:13:13 mypc kernel: [    8.090590] RIP: 0033:0x7f842612a3fd
Feb  1 23:13:13 mypc kernel: [    8.090592] Code: 28 89 54 24 1c 48 89 74 24 10 89 7c 24 08 e8 fa a4 f6 ff 8b 54 24 1c 48 8b 74 24 10 41 89 c0 8b 7c 24 08 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 33 44 89 c7 48 89 44 24 08 e8 3e a5 f6 ff 48
Feb  1 23:13:13 mypc kernel: [    8.090593] RSP: 002b:00007ffe82ca9510 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
Feb  1 23:13:13 mypc kernel: [    8.090594] RAX: ffffffffffffffda RBX: 000000000000000c RCX: 00007f842612a3fd
Feb  1 23:13:13 mypc kernel: [    8.090595] RDX: 0000000000000000 RSI: 00007ffe82ca9550 RDI: 000000000000000c
Feb  1 23:13:13 mypc kernel: [    8.090595] RBP: 000055bf626d4030 R08: 0000000000000000 R09: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.090596] R10: 0000000000000000 R11: 0000000000000293 R12: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.090596] R13: 00007ffe82ca96a0 R14: 00007ffe82ca969c R15: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.090597]  </TASK>
Feb  1 23:13:13 mypc kernel: [    8.090598] Modules linked in: vboxdrv(OE) intel_rapl_msr intel_rapl_common bnep snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_usb_audio snd_hda_codec snd_usbmidi_lib snd_hda_core edac_mce_amd binfmt_misc uvcvideo iwlmvm snd_hwdep snd_seq_midi videobuf2_vmalloc btusb kvm_amd snd_seq_midi_event videobuf2_memops videobuf2_v4l2 mac80211 snd_rawmidi eeepc_wmi btrtl videobuf2_common kvm snd_seq libarc4 videodev snd_pcm asus_wmi btbcm snd_seq_device platform_profile btintel sparse_keymap nls_iso8859_1 mc joydev input_leds corsair_cpro rapl snd_timer video bluetooth iwlwifi wmi_bmof efi_pstore snd ucsi_ccg typec_ucsi ecdh_generic ecc k10temp ccp typec soundcore cfg80211 mac_hid sch_fq_codel msr parport_pc ppdev lp parport sunrpc ip_tables x_tables autofs4 dm_crypt raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear system76_io(OE)
Feb  1 23:13:13 mypc kernel: [    8.090620]  system76_acpi(OE) hid_generic uas usbhid usb_storage hid amdgpu iommu_v2 gpu_sched drm_ttm_helper mxm_wmi ttm crct10dif_pclmul drm_kms_helper crc32_pclmul ghash_clmulni_intel syscopyarea sysfillrect sysimgblt aesni_intel fb_sys_fops igb atlantic cec crypto_simd nvme r8169 rc_core ahci dca cryptd xhci_pci drm i2c_piix4 macsec i2c_designware_pci libahci i2c_algo_bit xhci_pci_renesas realtek nvme_core wmi
Feb  1 23:13:13 mypc kernel: [    8.090631] ---[ end trace 96a2807216b1f270 ]---
[... 92 lines cut ...]
Feb  1 23:13:13 mypc kernel: [    8.091312]  do_syscall_64+0x5c/0xc0
Feb  1 23:13:13 mypc kernel: [    8.091313]  ? do_syscall_64+0x69/0xc0
Feb  1 23:13:13 mypc kernel: [    8.091314]  ? asm_exc_page_fault+0x8/0x30
Feb  1 23:13:13 mypc kernel: [    8.091315]  entry_SYSCALL_64_after_hwframe+0x44/0xae
Feb  1 23:13:13 mypc kernel: [    8.091317] RIP: 0033:0x7f842612a3fd
Feb  1 23:13:13 mypc kernel: [    8.091318] Code: 28 89 54 24 1c 48 89 74 24 10 89 7c 24 08 e8 fa a4 f6 ff 8b 54 24 1c 48 8b 74 24 10 41 89 c0 8b 7c 24 08 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 33 44 89 c7 48 89 44 24 08 e8 3e a5 f6 ff 48
Feb  1 23:13:13 mypc kernel: [    8.091318] RSP: 002b:00007ffe82ca9510 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
Feb  1 23:13:13 mypc kernel: [    8.091319] RAX: ffffffffffffffda RBX: 000000000000000c RCX: 00007f842612a3fd
Feb  1 23:13:13 mypc kernel: [    8.091320] RDX: 0000000000000000 RSI: 00007ffe82ca9550 RDI: 000000000000000c
Feb  1 23:13:13 mypc kernel: [    8.091320] RBP: 000055bf626d4030 R08: 0000000000000000 R09: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.091321] R10: 0000000000000000 R11: 0000000000000293 R12: 0000000000000000
[... cut ...]
Feb  1 23:13:13 mypc kernel: [    8.239627] R10: 0000000000000000 R11: 000000000000003f R12: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.239628] R13: ffff96f1a15b6980 R14: 0000000000000001 R15: 0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.239629] FS:  00007f8425114140(0000) GS:ffff97006eec0000(0000) knlGS:0000000000000000
Feb  1 23:13:13 mypc kernel: [    8.239630] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb  1 23:13:13 mypc kernel: [    8.239631] CR2: 00007f97ec008008 CR3: 0000000109a5c000 CR4: 0000000000750ee0

eitch, 2022-02-01:
After blacklisting the module, and then running:

    $ uname -a
    Linux mypc 5.16.5 #10 SMP PREEMPT Tue Feb 1 23:23:25 CET 2022 x86_64 x86_64 x86_64 GNU/Linux

I got the following error when modprobing the module:

	Feb  1 23:36:30 mypc kernel: [   56.933108] atlantic 0000:0f:00.0 enp15s0: renamed from eth0
	Feb  1 23:36:31 mypc kernel: [   57.434994] invalid opcode: 0000 [#1] PREEMPT SMP NOPTI
	Feb  1 23:36:31 mypc kernel: [   57.434999] CPU: 25 PID: 1370 Comm: NetworkManager Tainted: G           OE     5.16.5-eitch #10
	Feb  1 23:36:31 mypc kernel: [   57.435001] Hardware name: ASUS System Product Name/ROG CROSSHAIR VIII HERO (WI-FI), BIOS 3801 07/30/2021
	Feb  1 23:36:31 mypc kernel: [   57.435002] RIP: 0010:aq_nic_start+0x37e/0x380 [atlantic]
	Feb  1 23:36:31 mypc kernel: [   57.435008] Code: e8 e7 c1 9d f2 41 89 c4 85 c0 0f 88 3f fd ff ff 41 8b 95 94 01 00 00 b8 01 00 00 00 c4 e2 69 f7 c0 41 09 85 f8 05 00 00 eb 81 <0f> 0b 0f 1f 44 00 00 55 48 89 e5 41 57 41 56 41 55 41 54 53 49 89
	Feb  1 23:36:31 mypc kernel: [   57.435009] RSP: 0018:ffffb89c82ce73f0 EFLAGS: 00010202
	Feb  1 23:36:31 mypc kernel: [   57.435010] RAX: 0000000000000008 RBX: 0000000000000008 RCX: 0000000000000010
	Feb  1 23:36:31 mypc kernel: [   57.435011] RDX: 0000000000000019 RSI: 0000000000005be8 RDI: ffff98993d01d428
	Feb  1 23:36:31 mypc kernel: [   57.435012] RBP: ffffb89c82ce7408 R08: 0000000000000001 R09: 0000000000000000
	Feb  1 23:36:31 mypc kernel: [   57.435013] R10: 0000000000001000 R11: fffff4968c3a8b80 R12: 0000000000000000
	Feb  1 23:36:31 mypc kernel: [   57.435013] R13: ffff98992a8de980 R14: 0000000000000000 R15: ffff98992a8de000
	Feb  1 23:36:31 mypc kernel: [   57.435014] FS:  00007f43c27b3140(0000) GS:ffff98a6ef040000(0000) knlGS:0000000000000000
	Feb  1 23:36:31 mypc kernel: [   57.435015] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
	Feb  1 23:36:31 mypc kernel: [   57.435015] CR2: 000055c0bae6b120 CR3: 0000000104030000 CR4: 0000000000750ee0
	Feb  1 23:36:31 mypc kernel: [   57.435016] PKRU: 55555554
	Feb  1 23:36:31 mypc kernel: [   57.435017] Call Trace:
	Feb  1 23:36:31 mypc kernel: [   57.435017]  <TASK>
	Feb  1 23:36:31 mypc kernel: [   57.435019]  aq_ndev_open+0x49/0x70 [atlantic]
	Feb  1 23:36:31 mypc kernel: [   57.435023]  __dev_open+0xef/0x1a0
	Feb  1 23:36:31 mypc kernel: [   57.435026]  __dev_change_flags+0x1a3/0x210
	Feb  1 23:36:31 mypc kernel: [   57.435028]  dev_change_flags+0x26/0x60
	Feb  1 23:36:31 mypc kernel: [   57.435029]  do_setlink+0x28e/0x10d0
	Feb  1 23:36:31 mypc kernel: [   57.435030]  ? __snmp6_fill_stats64.constprop.0+0xe5/0x110
	Feb  1 23:36:31 mypc kernel: [   57.435032]  ? __nla_validate_parse+0x5b/0xb90
	Feb  1 23:36:31 mypc kernel: [   57.435035]  __rtnl_newlink+0x605/0xa10
	Feb  1 23:36:31 mypc kernel: [   57.435037]  ? skb_queue_tail+0x47/0x50
	Feb  1 23:36:31 mypc kernel: [   57.435038]  ? sock_def_readable+0x50/0x90
	Feb  1 23:36:31 mypc kernel: [   57.435040]  ? __netlink_sendskb+0x6d/0x90
	Feb  1 23:36:31 mypc kernel: [   57.435041]  ? netlink_unicast+0x2f6/0x330
	Feb  1 23:36:31 mypc kernel: [   57.435042]  ? rtnl_getlink+0x391/0x410
	Feb  1 23:36:31 mypc kernel: [   57.435045]  ? kmem_cache_alloc_trace+0x1a6/0x320
	Feb  1 23:36:31 mypc kernel: [   57.435048]  rtnl_newlink+0x49/0x70
	Feb  1 23:36:31 mypc kernel: [   57.435049]  rtnetlink_rcv_msg+0x168/0x3a0
	Feb  1 23:36:31 mypc kernel: [   57.435050]  ? __inode_attach_wb+0x26/0x1e0
	Feb  1 23:36:31 mypc kernel: [   57.435052]  ? rtnl_calcit.isra.0+0x130/0x130
	Feb  1 23:36:31 mypc kernel: [   57.435054]  netlink_rcv_skb+0x55/0x100
	Feb  1 23:36:31 mypc kernel: [   57.435055]  rtnetlink_rcv+0x15/0x20
	Feb  1 23:36:31 mypc kernel: [   57.435056]  netlink_unicast+0x22c/0x330
	Feb  1 23:36:31 mypc kernel: [   57.435057]  netlink_sendmsg+0x243/0x480
	Feb  1 23:36:31 mypc kernel: [   57.435058]  sock_sendmsg+0x65/0x70
	Feb  1 23:36:31 mypc kernel: [   57.435060]  ____sys_sendmsg+0x24e/0x290
	Feb  1 23:36:31 mypc kernel: [   57.435061]  ? import_iovec+0x31/0x40
	Feb  1 23:36:31 mypc kernel: [   57.435063]  ? sendmsg_copy_msghdr+0x7b/0xa0
	Feb  1 23:36:31 mypc kernel: [   57.435064]  ___sys_sendmsg+0x81/0xc0
	Feb  1 23:36:31 mypc kernel: [   57.435065]  ? security_file_free+0x54/0x60
	Feb  1 23:36:31 mypc kernel: [   57.435067]  ? kmem_cache_free+0x10a/0x420
	Feb  1 23:36:31 mypc kernel: [   57.435068]  ? call_rcu+0xa8/0x2a0
	Feb  1 23:36:31 mypc kernel: [   57.435070]  ? __fget_light+0xa3/0x120
	Feb  1 23:36:31 mypc kernel: [   57.435072]  __sys_sendmsg+0x62/0xb0
	Feb  1 23:36:31 mypc kernel: [   57.435073]  ? __x64_sys_close+0x11/0x40
	Feb  1 23:36:31 mypc kernel: [   57.435074]  __x64_sys_sendmsg+0x1d/0x20

cail, 2022-02-03:
Thats a full dup of #32

closed 2022-02-03 by cail