Aquantia/AQtion #36: Atlantic AQtion driver installed on Ubuntu 20.04.3 for QNAP QXG-5G1T-111C - Connection drops

TheBudda, 2022-02-03:
I have a 3 Gbps connection up & down. QXG-5G1T-111C is hitting the speeds. My motherboard is a - ASUS Rog Strix Z370G, CPU - i7 8700k 6 core CPU. 

I use git clone https://github.com/Aquantia/AQtion.git and followed the README.txt file. Everything went ok, but blobs were missing with the final command: sudo make install. The output is below. The driver is installed with 5.13.0-28-generic and drops out 5 - 10 times a day, but only on ubuntu. Windows 10 & 11 is driver works fine!

W: Possible missing firmware /lib/firmware/mrvl/91B1.fw for module atlantic
W: Possible missing firmware /lib/firmware/mrvl/87B1.fw for module atlantic
W: Possible missing firmware /lib/firmware/mrvl/80B1.fw for module atlantic

05:00.0 Ethernet controller: Aquantia Corp. AQC111 NBase-T/IEEE 802.3bz Ethernet Controller [AQtion] (rev 02

[image]
)

Can you please help this is so annoying that the connection drops out? All drivers do the same on Linux. This issue even happens with a fresh install of ubuntu 20.04.3, the card is a QNAP QXG-5G1T-111C, work out of the box. But keeps dropping connection. This is why I have tried this driver

cail, 2022-02-03:
Warning about the firmware can be ignored.
What do you mean with "drops down"? It that link goes down and up?
Can you provide the output of
"sudo dmesg | grep -i atlantic"
?

TheBudda, 2022-02-03:
> 

The connection to the internet drops cuts off then comes back on,  I see the spinning on networks in the taskbar. It does not do this with the motherboard controller. Only with the QNAP QXG-5G1T-111C. The card is only a week old. Below is the info you asked for. I hope it helps. I am 63 years old been on the net for over 20 years, I am running my own OVH server ;)

sudo dmesg | grep -i atlantic
[    0.672138] atlantic: loading out-of-tree module taints kernel.
[    0.672237] atlantic: module verification failed: signature and/or required key missing - tainting kernel
[    0.673522] atlantic: rbl_enabled = 0 fw_ver = 0x301006a
[    1.174570] atlantic: Chip features = 0x4400017 fw_ver_actual = 0x301006a
[    1.206916] atlantic 0000:05:00.0 enp5s0: renamed from eth0
[    8.178326] atlantic: rbl_enabled = 0 fw_ver = 0x301006a
[   14.119211] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000
[ 3439.942629] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 5000 new 0
[ 3451.515826] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000
[ 3681.258619] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 5000 new 0
[ 3692.972821] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000
[ 7294.571499] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 5000 new 0
[ 7300.147206] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000
[ 7555.947626] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 5000 new 0
[ 7581.763694] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000
[11430.756363] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 5000 new 0
[11444.435340] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000
[15044.078919] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 5000 new 0
[15049.550066] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000
[15305.495095] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 5000 new 0
[15311.086846] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000

cail, 2022-02-03:
Thanks. From log it looks like a hardware issue.
It could be:
- because of cabling. What cable, length and quality you use? Oficially, rates > 1G needs cat6.
- because of the quality of the hardware. It could be overheating or other conditions.
- because of some conflicts or incompatibility with the link partner.

We can do nothing here, on a driver level. I would assume the same should happen on windows as well. may be you just don't notice that?

TheBudda, 2022-02-03:
> Thanks. From the log, it looks like a hardware issue. It could be:
> 
>     * because of cabling. What cable, length, and quality do you use? Officially, rates > 1G needs cat6.
> 
>     * because of the quality of the hardware. It could be overheating or other conditions.
> 
>     * because of some conflicts or incompatibility with the link partner.
> 
> 
> We can do nothing here, on a driver level. I would assume the same should happen on windows as well. Maybe you overlook that?

First, I would like to thank you for your time on this. No issues on Windows! The cabling is a new Ugreen Cat7 1 meter from the ZYXEL router to the PC. BUT! The cable from the modem to the router is 3 meters long. I will change that cable! 

Sapphire RX 580 nitro+ 8GB with the QXG-5G1T-111C fits under the GPU, with a gap of 7CM. I do not do any gaming, only streaming. I can not see it being a heating issue! The internet connection can run for hours then drops out. I have a 3 Gbps up and down; you can see I am hitting the correct speeds. Can you please confirm Cat7 is ok or should I change the cables?

[image]
n

cail, 2022-02-03:
Cat7 is for sure more than enough.
I can only think about some incompatibilities with the router. But again, I can't explain why it does not happen on windows.
One more proposal is to check with other link partner. Connecting to another 5G device and observe "dmesg" the same way to see the dropouts.

TheBudda, 2022-02-03:
> Cat7 is for sure more than enough. I can only think about some incompatibilities with the router. But again, I can't explain why it does not happen on windows. One more proposal is to check with [...]

@cail thank you for your time dude I really appreciate it! 

There is no ISP that gives you a really good router if it is free lol my ISP gave me a new Adtran modem with a Zyxel AX7501-B0 Dual-Band lol they are only £99 to buy on eBay. The guy said that the SFP+ mod can get hot leave the backdoor open on the router lol I should not have to do that when I am paying £99 per month. I will look into getting a good router soon

https://service-provider.zyxel.com/emea/en/products/wifi-mesh-system/mpro-mesh-wifi-gateways/ax7501-b0
Like I said thank you for your time. Keep safe out there dude ;- )

TheBudda, 2022-04-16:
@cail UPDATE!

I contacted QNAP and told them about the loss of connection. QNAP support was fantastic. I sent the old card back and QNAP sent me a completely new card - QNAP QXG-5G1T-111C. When I got the new card I installed a fresh install of ubuntu 20.04.4 the new card worked out of the box. But, within an hour the card started to it dropped the connection again

So, I tried to install the driver from the QNAP website would not compile/build. So, I installed the Aquantia/AQtion driver with git clone https://github.com/Aquantia/AQtion.git everything went great and the driver was installed. It's now been a couple of hours running and no drops on the connection up to now! My fingers are crossed!! 

I run your command sudo dmesg | grep -i atlantic it seems to have the same output

budda@twopills:~$ sudo dmesg | grep -i atlantic
[sudo] password for budda: 
[    0.685840] atlantic: loading out-of-tree module taints kernel.
[    0.685936] atlantic: module verification failed: signature and/or required key missing - tainting kernel
[    0.687207] atlantic: rbl_enabled = 0 fw_ver = 0x301006a
[    1.188448] atlantic: Chip features = 0x4400017 fw_ver_actual = 0x301006a
[    1.267666] atlantic 0000:05:00.0 enp5s0: renamed from eth0
[    4.692879] atlantic: rbl_enabled = 0 fw_ver = 0x301006a
[   10.673910] [aq_nic_update_link_status:218(enp5s0)]atlantic: link change old 0 new 5000
budda@twopills:~$

Let's see what happens over the new few days