Aquantia/AQtion #37: Solo10G SFP+ (Thunderbolt 3 Edition): does not wake up, need to plug/unplug

ffries, 2022-04-23:
Dear all,

First of all, thanks for the hard work.

My Solo10G SFP+ is connected to a Lenovo Thinkpad Thunderbolt 3 dock
When starting-up my GNU/Linux station, the Solo10G SFP+ is not detected.
I need to plug/unplug the Solo10G thunderbolt cable a number of times to force detection.

Is there a workaround as on the long run it may hinder the thunderbolt cable?
The best would be a command that would force detection and/or waking-up.

I am using Debian testing stock kernel, please let me know if I need to reinstall your kernel module.
The reason is that I am booting in secure mode and I wonder whether installing a custom module is possible.

My kernel is:
Linux x280 5.16.0-6-amd64 #1 SMP PREEMPT Debian 5.16.18-1 (2022-03-29) x86_64 GNU/Linux

Kind regards,
Kellogs

ffries, 2022-04-24:
I installed the latest driver from git and the problem remains. The thunderbolt network card is not detected, as shows ifconfig -a:
```
ifconfig -a
enp0s31f6: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
enx083a8854e6af: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
virbr0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
wlp59s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
```

ffries, 2022-04-24:
sudo insmod -f /lib/modules/5.16.0-6-amd64/aquantia/atlantic.ko
insmod: ERROR: could not insert module /lib/modules/5.16.0-6-amd64/aquantia/atlantic.ko: Operation not permitted
~/Documents/logiciels/Aquantia/AQtion$ lsmod | grep atlantic

ffries, 2022-04-25:
Ok, I understand what is going on: as the system is protected with UEFI I cannot insert a module unless it is signed.
So I went back to the stock Debian kernel and linux atlantic module and the problems remains.
The network card is seen by the system but nothing happens:

```
boltctl list
 ● Lenovo ThinkPad Thunderbolt 3 Dock
   ├─ type:          peripheral
   ├─ name:          ThinkPad Thunderbolt 3 Dock
   ├─ vendor:        Lenovo
   ├─ uuid:          00e9298d-7a6f-0801-ffff-ffffffffffff
   ├─ generation:    Thunderbolt 3
   ├─ status:        authorized
   │  ├─ domain:     cb010000-0072-7098-a00a-271482258221
   │  ├─ rx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  ├─ tx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  └─ authflags:  boot
   ├─ authorized:    lun. 25 avril 2022 06:59:56
   ├─ connected:     lun. 25 avril 2022 06:59:56
   └─ stored:        mar. 08 mars 2022 07:19:31
      ├─ policy:     auto
      └─ key:        yes

 ● Sonnet Technologies, Inc Solo 10G SFP+ Thunderbolt 3 Edition
   ├─ type:          peripheral
   ├─ name:          Solo 10G SFP+ Thunderbolt 3 Edition
   ├─ vendor:        Sonnet Technologies, Inc
   ├─ uuid:          00b8607c-0b4c-0800-ffff-ffffffffffff
   ├─ generation:    Thunderbolt 3
   ├─ status:        authorized
   │  ├─ domain:     cb010000-0072-7098-a00a-271482258221
   │  ├─ rx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  ├─ tx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  └─ authflags:  boot
   ├─ authorized:    lun. 25 avril 2022 06:59:57
   ├─ connected:     lun. 25 avril 2022 06:59:57
   └─ stored:        dim. 24 avril 2022 07:37:20
      ├─ policy:     auto
      └─ key:        yes
```

cail, 2022-04-25:
ffries, you are listing your thunderbolt devices, not AQC PCI device.
Please do 'lspci' to check if AQC is there.
Your device should be a thunderbolt-pci bridge, eventually showing up with pci AQC device.
I would assume something is wrong with your kernel's thunderbolt stack.

ffries, 2022-04-25:
Thanks. Here us the output of lspci, no trace of AQC here.
```
lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 08)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.2 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #3 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point LPC Controller/eSPI Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (4) I219-LM (rev 21)
02:00.0 PCI bridge: Intel Corporation JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016] (rev 01)
03:00.0 PCI bridge: Intel Corporation JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016] (rev 01)
03:01.0 PCI bridge: Intel Corporation JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016] (rev 01)
03:02.0 PCI bridge: Intel Corporation JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016] (rev 01)
04:00.0 System peripheral: Intel Corporation JHL6240 Thunderbolt 3 NHI (Low Power) [Alpine Ridge LP 2016] (rev 01)
05:00.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge DD 2018] (rev 06)
06:02.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge DD 2018] (rev 06)
06:04.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge DD 2018] (rev 06)
07:00.0 USB controller: Intel Corporation JHL7540 Thunderbolt 3 USB Controller [Titan Ridge DD 2018] (rev 06)
08:00.0 PCI bridge: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] (rev 02)
09:01.0 PCI bridge: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] (rev 02)
3a:00.0 USB controller: Intel Corporation JHL6240 Thunderbolt 3 USB 3.1 Controller (Low Power) [Alpine Ridge LP 2016] (rev 01)
3b:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)
3c:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983
3d:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
```

ffries, 2022-04-25:
After unpluging/repluging, the Aquantia PCI device is visible:
`0a:00.0 Ethernet controller: Aquantia Corp. Device 80b1 (rev 02)`

ffries, 2022-04-26:
I modified my Thunderbolt setting. Now devices are not attached in pre-boot and only enabled with secure connection. The network card is seen by bolt utility but it is not listed in the PCI list. Any idea how to solve this?

josharmour, 2022-10-01:
Is there a fix to this issue?  I'm also seeing it with Ubuntu 22.04 LTS and the Sabrent th-s3ea adapter.  After a long period of being online or after a reboot I need to unplug and replug in the adapter to get ip connectivity.

ffries, 2023-11-14:
The issue is still present besides BIOS firmware updates.
One fix is to run:
sudo su
echo 1 > /sys/bus/pci/rescan
Then unplug and replug

Then the SPF+ led lights up and network connection is established.
There must be something simple that I don't understand and prevent the SPF+ port from waking-up.

ffries, 2023-12-27:
Unfortunately this issue is still present in latest Debian Bookworm.
But now I only need to unplug and replug to work in 100% of times.

I suspect an hibernation problem:
```
[   56.358720] hid-generic 0003:17EF:30CF.0005: hiddev0,hidraw0: USB HID v1.11 Device [Lenovo ThinkPad Thunderbolt 3 Dock USB Audio] on usb-0000:07:00.0-2.1.1.2/input3
[   86.973415] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[   93.364326] pcieport 0000:03:01.0: pciehp: Slot(1): Link Down
[   93.364331] pcieport 0000:03:01.0: pciehp: Slot(1): Card not present
[   93.364338] pcieport 0000:06:04.0: Unable to change power state from D3hot to D0, device inaccessible
[   93.364360] pcieport 0000:08:00.0: Unable to change power state from D3hot to D0, device inaccessible
[   93.364368] pcieport 0000:09:01.0: Unable to change power state from D3hot to D0, device inaccessible
[   93.364687] pcieport 0000:06:04.0: Runtime PM usage count underflow!
```