Aquantia/AQtion #38: Out of bounds error

rybak-i, 2022-04-30:
Bugs in aq_nic.c on lines:
329
592
652
1088
1570

++i, aq_vec = self->aq_vec[i]

aq_vec = self->aq_vec[i] should be put inside the loop.

Thanks

cail, 2022-05-02:
Hello, thanks, this was fixed in upstream already, will backport here later,

Thanks

koo5, 2022-05-03:
fwiw this is in stock ubuntu jammy:
```
koom@jj ~> sudo dmesg
[    0.000000] Linux version 5.15.0-27-generic (buildd@ubuntu) (gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #28-Ubuntu SMP Thu Apr 14 04:55:28 UTC 2022 (Ubuntu 5.15.0-27.28-generic 5.15.30)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-27-generic root=UUID=c7c571f7-ec4b-40b5-93d1-05f2c735e103 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009cfffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009d00000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a20b000-0x000000000affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000b000000-0x000000000b01ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000b020000-0x00000000da5bbfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000da5bc000-0x00000000dbaadfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dbaae000-0x00000000dbb2afff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dbb2b000-0x00000000dbfc6fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dbfc7000-0x00000000dcc8bfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dcc8c000-0x00000000deffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000df000000-0x00000000dfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041f37ffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xd6238018-0xd6258657] usable ==> usable
[    0.000000] e820: update [mem 0xd6238018-0xd6258657] usable ==> usable
[    0.000000] e820: update [mem 0xd6227018-0xd6237a57] usable ==> usable
[    0.000000] e820: update [mem 0xd6227018-0xd6237a57] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009cfffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009d00000-0x0000000009ffffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] reserve setup_data: [mem 0x000000000a200000-0x000000000a20afff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000000a20b000-0x000000000affffff] usable
[    0.000000] reserve setup_data: [mem 0x000000000b000000-0x000000000b01ffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000b020000-0x00000000d6227017] usable
[    0.000000] reserve setup_data: [mem 0x00000000d6227018-0x00000000d6237a57] usable
[    0.000000] reserve setup_data: [mem 0x00000000d6237a58-0x00000000d6238017] usable
[    0.000000] reserve setup_data: [mem 0x00000000d6238018-0x00000000d6258657] usable
[    0.000000] reserve setup_data: [mem 0x00000000d6258658-0x00000000da5bbfff] usable
[    0.000000] reserve setup_data: [mem 0x00000000da5bc000-0x00000000dbaadfff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000dbaae000-0x00000000dbb2afff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000dbb2b000-0x00000000dbfc6fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000dbfc7000-0x00000000dcc8bfff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000dcc8c000-0x00000000deffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000df000000-0x00000000dfffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fd000000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000041f37ffff] usable
[    0.000000] efi: EFI v2.70 by American Megatrends
[    0.000000] efi: ACPI 2.0=0xdbac7000 ACPI=0xdbac7000 SMBIOS=0xdcac2000 SMBIOS 3.0=0xdcac1000 MEMATTR=0xd6e6b698 ESRT=0xd95c9418 MOKvar=0xd62ef000 RNG=0xdcadba98 
[    0.000000] efi: seeding entropy pool
[    0.000000] random: fast init done
[    0.000000] secureboot: Secure boot disabled
[    0.000000] SMBIOS 3.2.1 present.
[... 1209 lines cut ...]
[  382.944432] RBP: 00007f83542115e5 R08: 0000000000000000 R09: 0000000000000001
[  382.944434] R10: 0000000000000000 R11: 0000000000000293 R12: 0000000000000013
[... cut ...]
[  386.694629] lvm2-activation-generator: lvmconfig failed
[  387.102424] lvm2-activation-generator: lvmconfig failed
[  387.373835] lvm2-activation-generator: lvmconfig failed
[  387.986224] lvm2-activation-generator: lvmconfig failed
[  388.250914] lvm2-activation-generator: lvmconfig failed
[  388.761535] lvm2-activation-generator: lvmconfig failed
[  389.039160] lvm2-activation-generator: lvmconfig failed
[  389.396585] atlantic 0000:06:00.0 enp6s0: atlantic: link change old 0 new 1000
koom@jj ~> 
```

cail, 2022-05-03:
That was just recently reached 5.15, probably will be in next maintenance release of it.