Aquantia/AQtion #39: It won’t build with kernel `5.18`

andrejpodzimek, 2022-06-05:
There are (at least) two errors:

```
/var/lib/dkms/atlantic/2.5.3/build/aq_sysfs.c:65:10: error: ‘struct kobj_type’ has no member named ‘default_attrs’
   65 |         .default_attrs = bar_attrs,
      |          ^~~~~~~~~~~~~
/var/lib/dkms/atlantic/2.5.3/build/aq_sysfs.c:65:26: error: initialization of ‘const struct sysfs_ops *’ from incompatible pointer type ‘struct attribute **’ [-Werror=incompatible-pointer-types]
   65 |         .default_attrs = bar_attrs,
      |                          ^~~~~~~~~
/var/lib/dkms/atlantic/2.5.3/build/aq_sysfs.c:65:26: note: (near initialization for ‘bar_type.sysfs_ops’)
/var/lib/dkms/atlantic/2.5.3/build/aq_sysfs.c:211:10: error: ‘struct kobj_type’ has no member named ‘default_attrs’
  211 |         .default_attrs = memreg_attrs,
      |          ^~~~~~~~~~~~~
/var/lib/dkms/atlantic/2.5.3/build/aq_sysfs.c:211:26: error: initialization of ‘const struct sysfs_ops *’ from incompatible pointer type ‘struct attribute **’ [-Werror=incompatible-pointer-types]
  211 |         .default_attrs = memreg_attrs,
      |                          ^~~~~~~~~~~~
/var/lib/dkms/atlantic/2.5.3/build/aq_sysfs.c:211:26: note: (near initialization for ‘memreg_type.sysfs_ops’)
/var/lib/dkms/atlantic/2.5.3/build/aq_macsec.c: In function ‘aq_set_txsc’:
/var/lib/dkms/atlantic/2.5.3/build/aq_macsec.c:332:63: warning: passing argument 2 of ‘aq_ether_addr_to_mac’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  332 |         aq_ether_addr_to_mac(tx_class_rec.mac_sa, secy->netdev->dev_addr);
      |                                                   ~~~~~~~~~~~~^~~~~~~~~~
/var/lib/dkms/atlantic/2.5.3/build/aq_macsec.c:39:61: note: expected ‘unsigned char *’ but argument is of type ‘const unsigned char *’
   39 | static void aq_ether_addr_to_mac(u32 mac[2], unsigned char *emac)
      |                                              ~~~~~~~~~~~~~~~^~~~
```

```
/var/lib/dkms/atlantic/2.5.3/build/aq_pci_func.c: In function ‘aq_pci_func_init’:
/var/lib/dkms/atlantic/2.5.3/build/aq_pci_func.c:133:15: error: implicit declaration of function ‘pci_set_dma_mask’ [-Werror=implicit-function-declaration]
  133 |         err = pci_set_dma_mask(pdev, DMA_BIT_MASK(64));
      |               ^~~~~~~~~~~~~~~~
/var/lib/dkms/atlantic/2.5.3/build/aq_pci_func.c:135:23: error: implicit declaration of function ‘pci_set_consistent_dma_mask’ [-Werror=implicit-function-declaration]
  135 |                 err = pci_set_consistent_dma_mask(pdev, DMA_BIT_MASK(64));
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

josharmour, 2022-10-01:
I'm getting the same issue.

```

/var/build_aq_drv/1664641587/Linux/aq_pci_func.c:133:15: error: implicit declaration of function ‘pci_set_dma_mask’ [-Werror=implicit-function-declaration]
  133 |         err = pci_set_dma_mask(pdev, DMA_BIT_MASK(64));

```
>

cail, 2022-10-04:
Please check v2.5.5 - should be fixed.

josharmour, 2022-10-05:
I'm able to make now but I can't load the module.

joshu@server:~/AQtion$ sudo make
[sudo] password for joshu: 
make -j4 -C /lib/modules/5.19.11-051911-generic/build M="/home/joshu/AQtion" modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.11-051911-generic'
  CC [M]  /home/joshu/AQtion/aq_main.o
  CC [M]  /home/joshu/AQtion/aq_nic.o
  CC [M]  /home/joshu/AQtion/aq_pci_func.o
  CC [M]  /home/joshu/AQtion/aq_vec.o
  CC [M]  /home/joshu/AQtion/aq_ring.o
  CC [M]  /home/joshu/AQtion/aq_ptp.o
  CC [M]  /home/joshu/AQtion/aq_filters.o
  CC [M]  /home/joshu/AQtion/aq_hw_utils.o
  CC [M]  /home/joshu/AQtion/aq_ethtool.o
  CC [M]  /home/joshu/AQtion/aq_drvinfo.o
  CC [M]  /home/joshu/AQtion/aq_trace.o
  CC [M]  /home/joshu/AQtion/aq_phy.o
  CC [M]  /home/joshu/AQtion/aq_compat.o
  CC [M]  /home/joshu/AQtion/aq_tsn.o
  CC [M]  /home/joshu/AQtion/aq_sysfs.o
  CC [M]  /home/joshu/AQtion/hw_atl/hw_atl_a0.o
  CC [M]  /home/joshu/AQtion/hw_atl/hw_atl_b0.o
  CC [M]  /home/joshu/AQtion/hw_atl/hw_atl_utils.o
  CC [M]  /home/joshu/AQtion/hw_atl/hw_atl_utils_fw2x.o
  CC [M]  /home/joshu/AQtion/hw_atl/hw_atl_llh.o
  CC [M]  /home/joshu/AQtion/hw_atl/hw_atl_fw_hostboot.o
  CC [M]  /home/joshu/AQtion/hw_atl/hw_atl_fw_image.o
  CC [M]  /home/joshu/AQtion/hw_atl2/hw_atl2_llh.o
  CC [M]  /home/joshu/AQtion/hw_atl2/hw_atl2.o
  CC [M]  /home/joshu/AQtion/hw_atl2/hw_atl2_utils.o
  CC [M]  /home/joshu/AQtion/hw_atl2/hw_atl2_utils_fw.o
  CC [M]  /home/joshu/AQtion/hw_atl2/hw_atl2_fw_hostboot.o
  CC [M]  /home/joshu/AQtion/macsec/macsec_api.o
  CC [M]  /home/joshu/AQtion/aq_dash.o
  CC [M]  /home/joshu/AQtion/aq_macsec.o
  LD [M]  /home/joshu/AQtion/atlantic.o
  MODPOST /home/joshu/AQtion/Module.symvers
  CC [M]  /home/joshu/AQtion/atlantic.mod.o
  LD [M]  /home/joshu/AQtion/atlantic.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.19.11-051911-generic'
joshu@server:~/AQtion$ sudo rmmod atlantic
rmmod: ERROR: Module atlantic is not currently loaded
joshu@server:~/AQtion$ sudo make load
modprobe ptp
modprobe crc_itu_t
modprobe -q macsec || true
insmod ./atlantic.ko
insmod: ERROR: could not insert module ./atlantic.ko: Invalid module format
make: *** [Makefile:49: load] Error 1
joshu@server:~/AQtion$

cail, 2022-10-06:
Something is wrong with your build or your kernel config. please check dmesg for any other errors/hints. I'm closing this ticket.

closed 2022-10-06 by cail