Aquantia/AQtion #40: Network statistics update interval

AlTech98, 2022-06-20:
If i try to read `/sys/class/net/enp7s0/statistics/rx_bytes` it give the same value if called multiple times in the same second.

Is there a way to change the statistics update interval?

I'm using:
- `Ethernet controller [0200]: Aquantia Corp. AQC107 NBase-T/IEEE 802.3bz Ethernet Controller [AQtion] [1d6a:d107] (rev 02)`
- Linux 5.18.3
- Manjaro

cail, 2022-10-04:
Hello,

Only through the code changes - but thats not recommended.

Stats update is a bulky operation. Why do you need under-second intervals?

zen2, 2026-11-15:
I got the same initial question because the behaviour is really strange:
- /proc/net/dev is updated once a second
- ethtool stats for InPackets, OutPackets, InOctets, OutOctets are updated once a second
- ethtool stats for InPacketsDma, OutPacketsDma, InOctetsDma, OutOctetsDma  are update several times per second
- all other network devices report several times per second

As a result, the monitoring tools and/or widgets display unusual results:
- a transmission speed curve that is a sawtooth shape or resembles a square wave
- reported speeds that jump sequentially from 0 to twice the bandwidth

(Kernel 7.3.4, Driver atlantic, Device PCIE AQC113)