Aquantia/AQtion #44: net-misc/AQtion-2.5.5: too many arguments to function ‘netif_napi_add'

HougeLangley, 2023-01-07:
Full build log

```
[32m * [39;49;00mPackage:    net-misc/AQtion-2.5.5:0
[32m * [39;49;00mRepository: hougeapps-overlay
[32m * [39;49;00mMaintainer: chutzpah@gentoo.org
[32m * [39;49;00mUSE:        abi_x86_64 amd64 elibc_glibc kernel_linux userland_GNU
[32m * [39;49;00mFEATURES:   network-sandbox preserve-libs sandbox userpriv usersandbox
 [32m*[0m Determining the location of the kernel source code
 [32m*[0m Found kernel source directory:
 [32m*[0m     /usr/src/linux
 [32m*[0m Found sources for kernel version:
 [32m*[0m     6.1.1-lqx3
 [32m*[0m Checking for suitable kernel configuration options ...
 [31;01m*[0m   CONFIG_AQTION:	 should not be set. But it is.
[A[120C [34;01m[ [31;01m!![34;01m ][0m
 [33;01m*[0m Please check to make sure these options are set correctly.
 [33;01m*[0m Failure to do so may cause unexpected problems.
>>> Unpacking source...
>>> Unpacking AQtion-2.5.5.tar.gz to /var/tmp/portage/net-misc/AQtion-2.5.5/work
>>> Source unpacked in /var/tmp/portage/net-misc/AQtion-2.5.5/work
>>> Preparing source in /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45 ...
>>> Source configured.
>>> Compiling source in /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45 ...
ln: failed to create symbolic link 'Module.symvers': File exists
 [32m*[0m Preparing atlantic module
make -j48 HOSTCC=x86_64-pc-linux-gnu-gcc 'LDFLAGS=-m elf_x86_64' all 
make -j4 -C /usr/src/linux M="/var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45" modules
make[1]: Entering directory '/usr/src/linux-6.1.1-liquorix-r1'
make[1]: warning: -j4 forced in submake: resetting jobserver mode.
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Gentoo 12.2.1_p20221224 p7) 12.2.1 20221224
  You are using:           x86_64-pc-linux-gnu-gcc (Gentoo 12.2.1_p20221224 p7) 12.2.1 20221224
  CC [M]  /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_main.o
  CC [M]  /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_nic.o
  CC [M]  /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_pci_func.o
  CC [M]  /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_vec.o
[01m[K/var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_vec.c:[m[K In function ‘[01m[Kaq_vec_alloc[m[K’:
[01m[K/var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_vec.c:126:9:[m[K [01;31m[Kerror: [m[Ktoo many arguments to function ‘[01m[Knetif_napi_add[m[K’
  126 |         [01;31m[Knetif_napi_add[m[K(aq_nic_get_ndev(aq_nic), &self->napi,
      |         [01;31m[K^~~~~~~~~~~~~~[m[K
In file included from [01m[K./include/linux/etherdevice.h:21[m[K,
                 from [01m[K/var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_common.h:14[m[K,
                 from [01m[K/var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_vec.h:15[m[K,
                 from [01m[K/var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_vec.c:12[m[K:
[01m[K./include/linux/netdevice.h:2562:1:[m[K [01;36m[Knote: [m[Kdeclared here
 2562 | [01;36m[Knetif_napi_add[m[K(struct net_device *dev, struct napi_struct *napi,
      | [01;36m[K^~~~~~~~~~~~~~[m[K
make[2]: *** [scripts/Makefile.build:250: /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45/aq_vec.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1990: /var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45] Error 2
make[1]: Leaving directory '/usr/src/linux-6.1.1-liquorix-r1'
make: *** [Makefile:36: all] Error 2
 [31;01m*[0m ERROR: net-misc/AQtion-2.5.5::hougeapps-overlay failed (compile phase):
 [31;01m*[0m   emake failed
 [31;01m*[0m 
 [31;01m*[0m If you need support, post the output of `emerge --info '=net-misc/AQtion-2.5.5::hougeapps-overlay'`,
 [31;01m*[0m the complete build log and the output of `emerge -pqv '=net-misc/AQtion-2.5.5::hougeapps-overlay'`.
 [31;01m*[0m The complete build log is located at '/var/tmp/portage/net-misc/AQtion-2.5.5/temp/build.log'.
 [31;01m*[0m The ebuild environment file is located at '/var/tmp/portage/net-misc/AQtion-2.5.5/temp/environment'.
 [31;01m*[0m Working directory: '/var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45'
 [31;01m*[0m S: '/var/tmp/portage/net-misc/AQtion-2.5.5/work/AQtion-340d608726cbfa04b6046d74a362e788e1e17d45'
```

HougeLangley, 2023-01-07:
Gentoo emerge --info

```
Portage 3.0.43 (python 3.10.9-final-0, default/linux/amd64/17.1/desktop/systemd/merged-usr, gcc-12, glibc-2.36-r6, 6.1.1-lqx3 x86_64)
=================================================================
System uname: Linux-6.1.1-lqx3-x86_64-Intel-R-_Xeon-R-_CPU_E5-2696_v2_@_2.50GHz-with-glibc2.36
KiB Mem:    32086576 total,  30840992 free
KiB Swap:    9435132 total,   9435132 free
Head commit of repository gentoo: 88858ec6d2420300a0b71a2552c8182df626d182

Head commit of repository 4nykey: d95d6671a3ec372528f19047aaf52fb47f22fa2d

Head commit of repository gentoo-zh: d9c711059e8df9d5d532304764c8cdbda3013369

Timestamp of repository guru: Sat, 07 Jan 2023 15:02:41 +0000
Head commit of repository guru: 0e6d612c5455742ef0b612d883b21383873c0d99

Head commit of repository hougeapps-overlay: ecd40aa81864d73285f1b666e6f438057af4d320

sh bash 5.2_p15
ld GNU ld (Gentoo 2.39 p5) 2.39.0
ccache version 4.7.4 [disabled]
app-misc/pax-utils:        1.3.5::gentoo
app-shells/bash:           5.2_p15::gentoo
dev-lang/perl:             5.36.0-r1::gentoo
dev-lang/python:           3.10.9::gentoo, 3.11.1::gentoo
dev-lang/rust:             1.66.0::gentoo
dev-util/ccache:           4.7.4::gentoo
dev-util/cmake:            3.25.1::gentoo
dev-util/meson:            1.0.0::gentoo
sys-apps/baselayout:       2.9::gentoo
sys-apps/sandbox:          2.29::gentoo
sys-apps/systemd:          252.4::gentoo
sys-devel/autoconf:        2.71-r5::gentoo
sys-devel/automake:        1.16.5::gentoo
sys-devel/binutils:        2.39-r4::gentoo
sys-devel/binutils-config: 5.4.1::gentoo
sys-devel/gcc:             12.2.1_p20221224::gentoo
sys-devel/gcc-config:      2.9::gentoo
sys-devel/libtool:         2.4.7-r1::gentoo
sys-devel/llvm:            15.0.6::gentoo
sys-devel/make:            4.4::gentoo
sys-kernel/linux-headers:  6.1::gentoo (virtual/os-headers)
sys-libs/glibc:            2.36-r6::gentoo
Repositories:

gentoo
    location: /var/db/repos/gentoo
    sync-type: git
    sync-uri: https://github.com/gentoo/gentoo.git
    priority: -1000
    volatile: True

4nykey
    location: /var/db/repos/4nykey
    sync-type: git
    sync-uri: https://github.com/4nykey/4nykey.git
    masters: gentoo
    volatile: True

gentoo-zh
    location: /var/db/repos/gentoo-zh
    sync-type: git
    sync-uri: https://github.com/microcai/gentoo-zh.git
    masters: gentoo
    volatile: True

guru
    location: /var/db/repos/guru
    sync-type: git
[... 15 lines cut ...]
CHOST="x86_64-pc-linux-gnu"
CONFIG_PROTECT="/etc /usr/share/gnupg/qualified.txt"
CONFIG_PROTECT_MASK="/etc/ca-certificates.conf /etc/dconf /etc/env.d /etc/fonts/fonts.conf /etc/gconf /etc/gentoo-release /etc/sandbox.d"
CXXFLAGS="-O2 -pipe"
DISTDIR="/var/cache/distfiles"
EMERGE_DEFAULT_OPTS="--keep-going --with-bdeps=y --autounmask-write=y --quiet-build=n --jobs=4 -l"
ENV_UNSET="CARGO_HOME DBUS_SESSION_BUS_ADDRESS DISPLAY GDK_PIXBUF_MODULE_FILE GOBIN GOPATH PERL5LIB PERL5OPT PERLPREFIX PERL_CORE PERL_MB_OPT PERL_MM_OPT XAUTHORITY XDG_CACHE_HOME XDG_CONFIG_HOME XDG_DATA_HOME XDG_RUNTIME_DIR XDG_STATE_HOME"
FCFLAGS="-O2 -pipe"
FEATURES="assume-digests binpkg-docompress binpkg-dostrip binpkg-logs binpkg-multi-instance buildpkg-live config-protect-if-modified distlocks ebuild-locks fixlafiles getbinpkg ipc-sandbox merge-sync multilib-strict network-sandbox news parallel-fetch pid-sandbox preserve-libs protect-owned qa-unresolved-soname-deps sandbox sfperms strict unknown-features-warn unmerge-logs unmerge-orphans userfetch userpriv usersandbox usersync xattr"
FFLAGS="-O2 -pipe"
GENTOO_MIRRORS="https://mirrors.ustc.edu.cn/gentoo/"
LANG="C.UTF8"
LDFLAGS="-Wl,-O1 -Wl,--as-needed"
LEX="flex"
LINGUAS="en_US zh_CN en zh"
MAKEOPTS="-j48"
PKGDIR="/var/cache/binpkgs"
PORTAGE_BINHOST="http://hougearch.litterhougelangley.club/gentoo/amd64/desktop/systemd/"
PORTAGE_CONFIGROOT="/"
PORTAGE_RSYNC_OPTS="--recursive --links --safe-links --perms --times --omit-dir-times --compress --force --whole-file --delete --stats --human-readable --timeout=180 --exclude=/distfiles --exclude=/local --exclude=/packages --exclude=/.git"
PORTAGE_TMPDIR="/var/tmp"
SHELL="/bin/fish"
USE="X a52 aac acl acpi alsa amd64 bluetooth branding bzip2 cairo cdda cdr cli crypt cups dbus dri dts dvd dvdr encode exif flac fortran gdbm gif gpm gtk gui iconv icu ipv6 jpeg lcms libglvnd libnotify libtirpc mad mng mp3 mp4 mpeg multilib ncurses nls nptl ogg opengl openmp pam pango pcre pdf png policykit ppds qt5 readline sdl seccomp sound spell ssl startup-notification svg systemd test-rust tiff truetype udev udisks unicode upower usb vorbis wxwidgets x264 xattr xcb xft xml xv xvid zlib" ABI_X86="64" ADA_TARGET="gnat_2021" APACHE2_MODULES="authn_core authz_core socache_shmcb unixd actions alias auth_basic authn_alias authn_anon authn_dbm authn_default authn_file authz_dbm authz_default authz_groupfile authz_host authz_owner authz_user autoindex cache cgi cgid dav dav_fs dav_lock deflate dir disk_cache env expires ext_filter file_cache filter headers include info log_config logio mem_cache mime mime_magic negotiation rewrite setenvif speling status unique_id userdir usertrack vhost_alias" CALLIGRA_FEATURES="karbon sheets words" COLLECTD_PLUGINS="df interface irq load memory rrdtool swap syslog" CPU_FLAGS_X86="mmx mmxext sse sse2" ELIBC="glibc" GPSD_PROTOCOLS="ashtech aivdm earthmate evermore fv18 garmin garmintxt gpsclock greis isync itrax mtk3301 nmea ntrip navcom oceanserver oldstyle oncore rtcm104v2 rtcm104v3 sirf skytraq superstar2 ti [... cut ...] geoip fuzzy condition tee tarpit sysrq proto steal rawnat logmark ipmark dhcpmac delude chaos account"
Unset:  ADDR2LINE, AR, ARFLAGS, AS, ASFLAGS, CC, CCLD, CONFIG_SHELL, CPP, CPPFLAGS, CTARGET, CXX, CXXFILT, ELFEDIT, EXTRA_ECONF, F77FLAGS, FC, GCOV, GPROF, INSTALL_MASK, LC_ALL, LD, LFLAGS, LIBTOOL, MAKE, MAKEFLAGS, NM, OBJCOPY, OBJDUMP, PORTAGE_BUNZIP2_COMMAND, PORTAGE_COMPRESS, PORTAGE_COMPRESS_FLAGS, PORTAGE_RSYNC_EXTRA_OPTS, RANLIB, READELF, RUSTFLAGS, SIZE, STRINGS, STRIP, YACC, YFLAGS
```

benedikt-bartscher, 2023-01-14:
same here on arch linux

qwrty-ftw, 2023-01-16:
Same on Debian 11

nixxholas, 2023-11-14:
Doesn't seem to fix the build issue after applying a patch manually to test https://github.com/Aquantia/AQtion/pull/48 out.

`make -j4 -C /lib/modules/6.2.0-36-generic/build M="/home/x/LinuxDriver/AQtion" modules
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-36-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  You are using:           gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  CC [M]  /home/x/LinuxDriver/AQtion/aq_main.o
  CC [M]  /home/x/LinuxDriver/AQtion/aq_nic.o
  CC [M]  /home/x/LinuxDriver/AQtion/aq_pci_func.o
  CC [M]  /home/x/LinuxDriver/AQtion/aq_vec.o
  CC [M]  /home/x/LinuxDriver/AQtion/aq_ring.o
  CC [M]  /home/x/LinuxDriver/AQtion/aq_ptp.o
  CC [M]  /home/x/LinuxDriver/AQtion/aq_filters.o
/home/x/LinuxDriver/AQtion/aq_ptp.c: In function ‘aq_ptp_init’:
/home/x/LinuxDriver/AQtion/aq_ptp.c:2130:9: error: too many arguments to function ‘netif_napi_add’
 2130 |         netif_napi_add(aq_nic_get_ndev(aq_nic), &aq_ptp->napi,
      |         ^~~~~~~~~~~~~~
In file included from ./include/linux/etherdevice.h:21,
                 from /home/x/LinuxDriver/AQtion/aq_common.h:14,
                 from /home/x/LinuxDriver/AQtion/aq_hw_utils.h:15,
                 from /home/x/LinuxDriver/AQtion/aq_ptp.c:19:
./include/linux/netdevice.h:2592:1: note: declared here
 2592 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:260: /home/x/LinuxDriver/AQtion/aq_ptp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2026: /home/x/LinuxDriver/AQtion] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-36-generic'
make: *** [Makefile:36: all] Error 2`

NickR23, 2023-12-17:
Was able to fix this on Debian kernel ver 6.1.0-16.

Had to remove all instances of `OCTEON_MGMT_NAPI_WEIGHT` from netif_napi_add (it appears in multiple files try `make` to find all of them) and remove all instances of `ctx->prepare` [netif_napi_add](https://github.com/torvalds/linux/commit/27418b55f094c3c8bb125759338f6476ab70f043).