Aquantia/AQtion #49: Kernel version of atlantic doesn't work with ASUS XC-100C v2 (based on AQC113CS), but this repo works

t3hk0d3, 2023-03-08:
Hey there,

I am using lastest linux kernel (v6.2.2), and device initialization fails with following error:

```
[    7.473717] atlantic 0000:03:00.0: enabling device (0000 -> 0002)
[    7.674201] atlantic: Boot code hanged
[    7.674370] atlantic: probe of 0000:03:00.0 failed with error -110
```

However after loading this version of driver (after fixing compilation errors) everything works:

```
[92939.969084] atlantic: loading out-of-tree module taints kernel.
[92939.969240] atlantic: module verification failed: signature and/or required key missing - tainting kernel
[92939.977050] atlantic 0000:03:00.0 (unnamed net_device) (uninitialized): Hardware revision 0x0
[92940.013760] atlantic: Detect ATL2FW 1030012
[92940.019446] atlantic 0000:03:00.0 enp3s0: renamed from eth0
```

However this has few issues:
- Kernel is tainted
- periodic debug output

Is there any way to fix AQC113CS support in mainline linux kernel?

cail, 2023-03-09:
Hi Igor, thanks for the analysis, we'll check this internally.
Something with timings in upstream version, or something was changed unexpectedly.
The error is for hardware init - when driver can't reset the device.

t3hk0d3, 2023-03-31:
@cail Thank you. Let me know if you need any more information or testing. I still have NIC laying around.