Aquantia/AQtion #62: AQC107 fails to resume from suspend

johndoe31415, 2024-07-22:
Hello there!

I'm using an AQC107 NIC:

```
01:00.0 Ethernet controller [0200]: Aquantia Corp. AQtion AQC107 NBase-T/IEEE 802.3an Ethernet Controller [Atlantic 10G] [1d6a:07b1] (rev 02)
```

on Linux x86_64 running a standard Ubuntu (2024.04 noble) stock kernel, untainted:

```
reliant joe [~]: uname -a
Linux reliant 6.5.0-28-generic #29-Ubuntu SMP PREEMPT_DYNAMIC Thu Mar 28 23:46:48 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

reliant joe [~]: cat /proc/sys/kernel/tainted                                  
0
```

I'm experiencing sporadic NIC failures when waking from suspend-to-RAM. Concretely, when it happens (maybe on every 3rd suspend operation, so reasonably/annoyingly often), the network driver completely locks up and no connectivity is possible. Sometimes I'm able to recover by `rmmod` and `modprobe`, but in 90% of the cases this also is not possible and I have to reboot to get the NIC working again. Note also that the system will go into shutdown but systemd then hangs somewhere, needing me to issue a hard reset.

When it occurs, I see the following in dmesg:

```
[33284.397291] kworker/u256:70: page allocation failure: order:5, mode:0x40d00(GFP_NOIO|__GFP_COMP|__GFP_ZERO), nodemask=(null),cpuset=/,mems_allowed=0
[33284.397303] CPU: 17 PID: 4159607 Comm: kworker/u256:70 Not tainted 6.5.0-28-generic #29-Ubuntu
[33284.397306] Hardware name: LENOVO 30E0003QGE/1046, BIOS S07KT4AA 07/22/2022
[33284.397308] Workqueue: events_unbound async_run_entry_fn
[33284.397313] Call Trace:
[33284.397315]  <TASK>
[33284.397319]  dump_stack_lvl+0x48/0x70
[33284.397323]  dump_stack+0x10/0x20
[33284.397325]  warn_alloc+0x174/0x1f0
[33284.397329]  ? __alloc_pages_direct_compact+0xb7/0x240
[33284.397334]  __alloc_pages_slowpath.constprop.0+0x8f1/0x980
[33284.397339]  __alloc_pages+0x31f/0x350
[33284.397344]  ? aq_ring_alloc+0x27/0x90 [atlantic]
[33284.397359]  __kmalloc_large_node+0x7a/0x150
[33284.397362]  ? iommu_dma_alloc+0x16e/0x1e0
[33284.397366]  __kmalloc+0xdb/0x170
[33284.397370]  aq_ring_alloc+0x27/0x90 [atlantic]
[33284.397383]  aq_ring_rx_alloc+0x97/0xb0 [atlantic]
[33284.397396]  aq_vec_ring_alloc+0xbe/0x290 [atlantic]
[33284.397409]  ? hw_atl_b0_hw_ring_rx_fill+0x5d/0x70 [atlantic]
[33284.397424]  aq_nic_init+0x13d/0x240 [atlantic]
[33284.397439]  atl_resume_common+0x46/0xf0 [atlantic]
[33284.397452]  aq_pm_resume_restore+0xe/0x20 [atlantic]
[33284.397465]  pci_pm_resume+0x75/0x110
[33284.397468]  ? __pfx_pci_pm_resume+0x10/0x10
[33284.397471]  dpm_run_callback+0x54/0x1b0
[33284.397475]  device_resume+0xad/0x220
[33284.397478]  async_resume+0x1f/0x90
[33284.397480]  async_run_entry_fn+0x33/0x130
[33284.397483]  process_one_work+0x223/0x440
[33284.397487]  worker_thread+0x4d/0x3f0
[33284.397490]  ? __pfx_worker_thread+0x10/0x10
[33284.397492]  kthread+0xf2/0x120
[33284.397495]  ? __pfx_kthread+0x10/0x10
[33284.397498]  ret_from_fork+0x47/0x70
[33284.397501]  ? __pfx_kthread+0x10/0x10
[33284.397504]  ret_from_fork_asm+0x1b/0x30
[33284.397510]  </TASK>
[33284.397511] Mem-Info:
[33284.397513] active_anon:1394551 inactive_anon:357689 isolated_anon:0
                active_file:1793872 inactive_file:9717634 isolated_file:0
                unevictable:104 dirty:42 writeback:0
                slab_reclaimable:1842731 slab_unreclaimable:234794
                mapped:327615 shmem:140848 pagetables:23776
                sec_pagetables:0 bounce:0
                kernel_misc_reclaimable:0
                free:292280 free_pcp:0 free_cma:0
[33284.397518] Node 0 active_anon:5578204kB inactive_anon:1430756kB active_file:7175488kB inactive_file:38870536kB unevictable:416kB isolated(anon):0kB isolated(file):0kB mapped:1310460kB dirty:168kB writeback:0kB shmem:563392kB shmem_thp: 0kB shmem_pmdmapped: 0kB anon_thp: 0kB writeback_tmp:0kB kernel_stack:46704kB pagetables:95104kB sec_pagetables:0kB all_unreclaimable? no
[33284.397523] Node 0 DMA free:11264kB boost:0kB min:12kB low:24kB high:36kB reserved_highatomic:0KB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:15996kB managed:15360kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[33284.397527] lowmem_reserve[]: 0 2858 64098 64098 64098
[33284.397532] Node 0 DMA32 free:262836kB boost:12876kB min:15796kB low:18628kB high:21460kB reserved_highatomic:2048KB active_anon:21740kB inactive_anon:3736kB active_file:308kB inactive_file:1991556kB unevictable:0kB writepending:0kB present:2992764kB managed:2926724kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[33284.397537] lowmem_reserve[]: 0 0 61240 61240 61240
[33284.397541] Node 0 Normal free:895020kB boost:285064kB min:349708kB low:412408kB high:475108kB reserved_highatomic:2048KB active_anon:5556464kB inactive_anon:1427020kB active_file:7175180kB inactive_file:36878980kB unevictable:416kB writepending:168kB present:63949824kB managed:62710260kB mlocked:416kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[33284.397546] lowmem_reserve[]: 0 0 0 0 0
[33284.397551] Node 0 DMA: 0*4kB 0*8kB 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 1*1024kB (U) 1*2048kB (M) 2*4096kB (M) = 11264kB
[33284.397564] Node 0 DMA32: 4253*4kB (UME) 2490*8kB (UME) 1727*16kB (UMEH) 2056*32kB (UMEH) 1180*64kB (UMEH) 443*128kB (UMEH) 1*256kB (H) 0*512kB 0*1024kB 0*2048kB 0*4096kB = 262836kB
[33284.397581] Node 0 Normal: 130947*4kB (UMEH) 44782*8kB (UMEH) 121*16kB (UMEH) 345*32kB (UME) 4*64kB (ME) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 895276kB
[33284.397596] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=1048576kB
[33284.397597] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=2048kB
[33284.397598] 11652354 total pagecache pages
[33284.397599] 0 pages in swap cache
[33284.397600] Free swap  = 0kB
[33284.397601] Total swap = 0kB
[33284.397602] 16739646 pages RAM
[33284.397603] 0 pages HighMem/MovableOnly
[33284.397603] 326560 pages reserved
[33284.397604] 0 pages hwpoisoned
[33284.403493] atlantic 0000:01:00.0: PM: dpm_run_callback(): pci_pm_resume+0x0/0x110 returns -12
[33284.403497] atlantic 0000:01:00.0: PM: failed to resume async: error -12
```

Any advice on how I can support debugging this issue is greatly appreciated. Thanks!

tarkh, 2024-12-12:
Same here with `Ethernet controller: Aquantia Corp. AQtion AQC107 NBase-T/IEEE 802.3an Ethernet Controller [Atlantic 10G] (rev 02)`. @johndoe31415 did you find any solution? Thanx!

johndoe31415, 2025-03-13:
Hey there, sorry I must have missed your reply entirely. I just came to check up on this and found it now. As you may have guessed, I'm still suffering from the issue.

Your workaround is pretty clever, I did not think to remove the driver before hibernation! Is the PCI bus rescan necessary, i did some fiddling and it seemed to auto-detect the NIC as soon as `modprobe` went through.

I'm still quite disappointed that such an issue is present and being ignored.

Thanks again! Cheers, Joe

perrin4869, 2025-03-13:
I have a similar issue, maybe it's related? I am using NetworkManager, and I actually am not currently connecting any devices to the network via this NIC (in my case, AQC113CS in my MSI MEG ACE X670E). Firmware version is 1.3.21.
After a few days of regular usage, and a few suspend/resume cycles (always more than 3), suspend will fail and I will find the following log messages:

```
Nov 22 19:04:38 landau NetworkManager[2051]: <info>  [1732269878.8653] manager: sleep: sleep requested (sleeping: no  enabled: yes)
Nov 22 19:04:38 landau NetworkManager[2051]: <info>  [1732269878.8654] device (eth0): state change: unavailable -> unmanaged (reason 'unmanaged-sleeping', managed-type: 'full')
Nov 22 19:04:43 landau elogind-daemon[1995]: Delay lock is active (UID 0/root, PID 2051/NetworkManager) but inhibitor timeout is reached.
Nov 22 19:04:43 landau e-suspend[30663]: Performing sleep operation 'suspend'...
Nov 22 19:04:43 landau kernel: PM: suspend entry (deep)
Nov 22 19:04:43 landau kernel: Filesystems sync: 0.013 seconds
Nov 22 19:05:04 landau kernel: Freezing user space processes
Nov 22 19:05:04 landau kernel: task:NetworkManager  state:D stack:0     pid:2051  ppid:1      flags:0x00004006
Nov 22 19:05:04 landau kernel: Call Trace:
Nov 22 19:05:04 landau kernel:  <TASK>
Nov 22 19:05:04 landau kernel:  __schedule+0x382/0x1430
Nov 22 19:05:04 landau kernel:  ? finish_task_switch.isra.0+0x95/0x2c0
Nov 22 19:05:04 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Nov 22 19:05:04 landau kernel:  ? ktime_get+0x38/0xa0
Nov 22 19:05:04 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Nov 22 19:05:04 landau kernel:  schedule+0x5d/0xe0
Nov 22 19:05:04 landau kernel:  schedule_hrtimeout_range_clock+0xe3/0x1c0
Nov 22 19:05:04 landau kernel:  ? __pfx_hrtimer_wakeup+0x10/0x10
Nov 22 19:05:04 landau kernel:  usleep_range_state+0x63/0x90
Nov 22 19:05:04 landau kernel:  napi_disable+0x62/0x70
Nov 22 19:05:04 landau kernel:  aq_nic_stop+0x142/0x190 [atlantic]
Nov 22 19:05:04 landau kernel:  aq_ndev_close+0x16/0x30 [atlantic]
Nov 22 19:05:04 landau kernel:  __dev_close_many+0x9e/0x120
Nov 22 19:05:04 landau kernel:  __dev_change_flags+0xda/0x210
Nov 22 19:05:04 landau kernel:  dev_change_flags+0x22/0x60
Nov 22 19:05:04 landau kernel:  do_setlink+0x36b/0x1240
Nov 22 19:05:04 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Nov 22 19:05:04 landau kernel:  ? update_load_avg+0x7e/0x760
Nov 22 19:05:04 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Nov 22 19:05:04 landau last message buffered 1 times
Nov 22 19:05:04 landau kernel:  ? sched_clock_cpu+0xd/0x1e0
Nov 22 19:05:04 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Nov 22 19:05:04 landau kernel:  ? __nla_validate_parse+0x5f/0xcd0
Nov 22 19:05:04 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Nov 22 19:05:04 landau last message buffered 1 times
Nov 22 19:05:04 landau kernel:  __rtnl_newlink+0x5b1/0xa90
Nov 22 19:05:04 landau kernel:  ? timerqueue_del+0x2a/0x70
Nov 22 19:05:04 landau kernel:  ? rtnl_newlink+0x2a/0x70
Nov 22 19:05:04 landau kernel:  rtnl_newlink+0x43/0x70
Nov 22 19:05:04 landau kernel:  rtnetlink_rcv_msg+0x13e/0x3d0
Nov 22 19:05:04 landau kernel:  ? __pfx_rtnetlink_rcv_msg+0x10/0x10
Nov 22 19:05:04 landau kernel:  netlink_rcv_skb+0x4c/0x100
Nov 22 19:05:04 landau kernel:  netlink_unicast+0x245/0x370
Nov 22 19:05:04 landau kernel:  netlink_sendmsg+0x250/0x4d0
Nov 22 19:05:04 landau kernel:  ____sys_sendmsg+0x312/0x380
Nov 22 19:05:04 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Nov 22 19:05:04 landau kernel:  ___sys_sendmsg+0x9a/0xe0
Nov 22 19:05:04 landau kernel:  __sys_sendmsg+0xc8/0xf0
Nov 22 19:05:04 landau kernel:  do_syscall_64+0x35/0x80
Nov 22 19:05:04 landau kernel:  entry_SYSCALL_64_after_hwframe+0x78/0xe2
Nov 22 19:05:04 landau kernel: RIP: 0033:0x7f42eaf24b0d
Nov 22 19:05:04 landau kernel: RSP: 002b:00007ffdc45cad50 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
Nov 22 19:05:04 landau kernel: RAX: ffffffffffffffda RBX: 000000000f9cd5c0 RCX: 00007f42eaf24b0d
Nov 22 19:05:04 landau kernel: RDX: 0000000000000000 RSI: 00007ffdc45cadc0 RDI: 000000000000000d
Nov 22 19:05:04 landau kernel: RBP: 00007ffdc45cadc0 R08: 0000000000000000 R09: 0000000000000000
Nov 22 19:05:04 landau kernel: R10: 0000000000000000 R11: 0000000000000293 R12: 000000000f9c4c80
Nov 22 19:05:04 landau kernel: R13: 0000000000000000 R14: 00007ffdc45caf38 R15: 00007ffdc45caf2c
Nov 22 19:05:04 landau kernel:  </TASK>
```

I confirmed the problem comes from the NIC because I disabled the NIC in the BIOS and could use the PC with suspend/resume for over a week without incidents.
I want to look into updating the firmware at some point, just in case the problem is there... but probably it's something about the driver?

perrin4869, 2025-03-16:
I just found extra log messages, I missed those in the past, of the NIC when it tries to resume:

```
Mar 16 21:19:03 landau kernel: Workqueue: async async_run_entry_fn
Mar 16 21:19:03 landau kernel: Call Trace:
Mar 16 21:19:03 landau kernel:  <TASK>
Mar 16 21:19:03 landau kernel:  dump_stack_lvl+0x4a/0x70
Mar 16 21:19:03 landau kernel:  warn_alloc+0x160/0x180
Mar 16 21:19:03 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Mar 16 21:19:03 landau kernel:  ? __alloc_pages_direct_compact+0xb0/0x1e0
Mar 16 21:19:03 landau kernel:  __alloc_pages_slowpath.constprop.0+0x975/0xdb0
Mar 16 21:19:03 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Mar 16 21:19:03 landau kernel:  ? aq_ring_alloc+0x22/0x80 [atlantic]
Mar 16 21:19:03 landau kernel:  __alloc_pages_noprof+0x312/0x330
Mar 16 21:19:03 landau kernel:  ? aq_ring_alloc+0x22/0x80 [atlantic]
Mar 16 21:19:03 landau kernel:  ___kmalloc_large_node+0x69/0x100
Mar 16 21:19:03 landau kernel:  __kmalloc_large_node_noprof+0x19/0xa0
Mar 16 21:19:03 landau kernel:  __kmalloc_noprof+0x2a8/0x3a0
Mar 16 21:19:03 landau kernel:  ? aq_ring_alloc+0x22/0x80 [atlantic]
Mar 16 21:19:03 landau kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Mar 16 21:19:03 landau kernel:  aq_ring_alloc+0x22/0x80 [atlantic]
Mar 16 21:19:03 landau kernel:  aq_vec_ring_alloc+0xfd/0x1d0 [atlantic]
Mar 16 21:19:03 landau kernel:  aq_nic_init+0x113/0x1d0 [atlantic]
Mar 16 21:19:03 landau kernel:  atl_resume_common+0x43/0xd0 [atlantic]
Mar 16 21:19:03 landau kernel:  ? __pfx_pci_pm_resume+0x10/0x10
Mar 16 21:19:03 landau kernel:  dpm_run_callback+0x1d/0xd0
Mar 16 21:19:03 landau kernel:  device_resume+0x9c/0x210
Mar 16 21:19:03 landau kernel:  async_resume+0x19/0x30
Mar 16 21:19:03 landau kernel:  async_run_entry_fn+0x23/0xb0
Mar 16 21:19:03 landau kernel:  process_one_work+0x168/0x320
Mar 16 21:19:03 landau kernel:  worker_thread+0x2fa/0x430
Mar 16 21:19:03 landau kernel:  ? __pfx_worker_thread+0x10/0x10
Mar 16 21:19:03 landau kernel:  kthread+0xda/0x110
Mar 16 21:19:03 landau kernel:  ? __pfx_kthread+0x10/0x10
Mar 16 21:19:03 landau kernel:  ret_from_fork+0x2d/0x50
Mar 16 21:19:03 landau kernel:  ? __pfx_kthread+0x10/0x10
Mar 16 21:19:03 landau kernel:  ret_from_fork_asm+0x1a/0x30
Mar 16 21:19:03 landau kernel:  </TASK>
Mar 16 21:19:03 landau kernel: Mem-Info:
Mar 16 21:19:03 landau kernel: active_anon:6439720 inactive_anon:16281 isolated_anon:0\x0a active_file:1321632 inactive_file:3562976 isolated_file:0\x0a unevictable:42 dirty:5 writeback:0\x0a slab_reclaimable:366358 slab_unreclaimable:72354\x0a mapped:405377 shmem:132774 pagetables:60064\x0a sec_pagetables:1896 bounce:0\x0a kernel_misc_reclaimable:0\x0a free:582921 free_pcp:0 free_cma:0
Mar 16 21:19:03 landau kernel: Node 0 active_anon:25758880kB inactive_anon:65124kB active_file:5286528kB inactive_file:14251904kB unevictable:168kB isolated(anon):0kB isolated(file):0kB mapped:1621508kB dirty:20kB writeback:0kB shmem:531096kB shmem_thp:0kB shmem_pmdmapped:0kB anon_thp:0kB writeback_tmp:0kB kernel_stack:62916kB pagetables:240256kB sec_pagetables:7584kB all_unreclaimable? no
Mar 16 21:19:03 landau kernel: Node 0 DMA free:1044kB boost:0kB min:16kB low:28kB high:40kB reserved_highatomic:0KB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:10224kB unevictable:0kB writepending:0kB present:15996kB managed:15364kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
Mar 16 21:19:03 landau kernel: lowmem_reserve[]: 0 2081 63344 0 0
Mar 16 21:19:03 landau kernel: Node 0 DMA32 free:673472kB boost:9708kB min:11924kB low:14052kB high:16180kB reserved_highatomic:0KB active_anon:204056kB inactive_anon:0kB active_file:428kB inactive_file:809792kB unevictable:0kB writepending:0kB present:2223480kB managed:2157332kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
Mar 16 21:19:03 landau kernel: lowmem_reserve[]: 0 0 61263 0 0
Mar 16 21:19:03 landau kernel: Node 0 Normal free:1657168kB boost:286216kB min:351564kB low:414296kB high:477028kB reserved_highatomic:2048KB active_anon:25554824kB inactive_anon:65124kB active_file:5286100kB inactive_file:13431888kB unevictable:168kB writepending:20kB present:63928832kB managed:62741184kB mlocked:168kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
Mar 16 21:19:03 landau kernel: lowmem_reserve[]: 0 0 0 0 0
Mar 16 21:19:03 landau kernel: Node 0 DMA: 5*4kB (UM) 0*8kB 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 1*1024kB (U) 0*2048kB 0*4096kB = 1044kB
Mar 16 21:19:03 landau kernel: Node 0 DMA32: 37602*4kB (UME) 24331*8kB (UME) 11600*16kB (UME) 4463*32kB (UME) 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 673472kB
Mar 16 21:19:03 landau kernel: Node 0 Normal: 191002*4kB (UMEH) 51408*8kB (UMEH) 18137*16kB (UMEH) 5828*32kB (UMEH) 5*64kB (EH) 1*128kB (H) 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 1652408kB
Mar 16 21:19:03 landau kernel: Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=1048576kB
Mar 16 21:19:03 landau kernel: Node 0 hugepages_total=5000 hugepages_free=5000 hugepages_surp=0 hugepages_size=2048kB
Mar 16 21:19:03 landau kernel: 5017284 total pagecache pages
Mar 16 21:19:03 landau kernel: 0 pages in swap cache
Mar 16 21:19:03 landau kernel: Free swap  = 64913876kB
Mar 16 21:19:03 landau kernel: Total swap = 64913876kB
Mar 16 21:19:03 landau kernel: 16542077 pages RAM
Mar 16 21:19:03 landau kernel: 0 pages HighMem/MovableOnly
Mar 16 21:19:03 landau kernel: 313607 pages reserved
[... cut ...]
Mar 16 21:19:03 landau kernel: atlantic 0000:07:00.0: PM: dpm_run_callback(): pci_pm_resume returns -12
Mar 16 21:19:03 landau kernel: atlantic 0000:07:00.0: PM: failed to resume async: error -12
```

The ones I posted the other day were logs of the NIC trying to suspend after resume has failed as the error above
The logs are very similar to @tarkh's logs, so I think the issue is the same
What is your operating system? @tarkh 
I am using slackware64-current
Did any kernel version ever work properly for you?

johndoe31415, 2025-04-01:
@perrin4869 I've not yet tried more recent kernel versions, I'm on Ubuntu on the moment and, aside from the awful AQC107 suspend drama, am fairly happy with that. It has not been bothering me enough to get a vanilla kernel and try to tinker a config that works on my machine.

I've been living with the workaround @tarkh posted fairly happily but, as I recently discovered, it has one major drawback: WOL obviously doesn't work anymore, since it rmmods the driver right before suspend-to-RAM.

I really just wish they would fix the driver or at least acknowledge the issue :-(

perrin4869, 2025-04-01:
I actually finally figured out the cause of the issue. It seems to be caused when there is high levels of memory fragmentation in the system. 
You can check the levels of fragmentation by running `cat /proc/buddyinfo`, which lists the number of free memory blocks by order. When the driver crashes, it always seems to happen when the amount of free blocks is low. In my case, it seems that Firefox is causing the fragmentation? By killing Firefox and running memory compaction (`echo 1 > /proc/sys/vm/compact_memory`) while monitoring the amount of free blocks (`watch -n 1 cat /proc/buddyinfo`), can wait for the blocks to compact before suspending, and prevent the driver from crashing.

Now, obviously the driver needs to handle memory allocation without crashing the system when it fails, so someone at marvell needs to acknowledge the issue...

perrin4869, 2025-04-02:
ok, I finally tracked down an issue tracking this: https://lore.kernel.org/netdev/CAHk-%3DwiZZi7FcvqVSUirHBjx0bBUZ4dFrMDVLc3%2B3HCrtq0rBA%40mail.gmail.com/T/

dylandn, 2026-04-05:
Just reporting my system also has this 'fails to wake after several cycles of sleep/wake' issue with an AQC113C on 6.18.16
05:00.0 Ethernet controller: Aquantia Corp. AQC113C NBase-T/IEEE 802.3an Ethernet Controller [Marvell Scalable mGig] (rev 03)