Aquantia/AQtion #69: Build issue AQtion-2.5.12 - error: ‘PCI_IRQ_LEGACY’ undeclared (first use in this function)

epox33d, 2024-12-30:
OS: Debian GNU/Linux 12 (bookworm) x86_64
Kernel: 6.13.0-rc5-custom #1 SMP PREEMPT_DYNAMIC Mon Dec 30 21:52:20 MSK 2024 x86_64 GNU/Linux
Network apadter with AQC113C @ pcie x4

Trying to bild driver 05-23-22_Marvell_Linux_2.5.5_1_1 downloaded from official Marvell(Aquantia) https://www.marvell.com/support/downloads.html
Got error: "PCI_IRQ_LEGACY’ undeclared (first use in this function)"

With stock kernel 6.1.0-28-amd64 build error "too many arguments to function ‘netif_napi_add" https://github.com/Aquantia/AQtion/issues/44

```
root@*****:/opt/07-18-24_Marvell_Linux_2.5.12/Linux# make
make -j4 -C /lib/modules/6.13.0-rc5-custom/build M="/opt/07-18-24_Marvell_Linux_2.5.12/Linux" modules
make[1]: Entering directory '/usr/src/linux-headers-6.13.0-rc5-custom'
make[2]: Entering directory '/opt/07-18-24_Marvell_Linux_2.5.12/Linux'
  CC [M]  aq_main.o
  CC [M]  aq_nic.o
  CC [M]  aq_pci_func.o
  CC [M]  aq_vec.o
aq_pci_func.c: In function ‘aq_pci_probe’:
aq_pci_func.c:383:37: error: ‘PCI_IRQ_LEGACY’ undeclared (first use in this function); did you mean ‘NR_IRQS_LEGACY’?
  383 |                                     PCI_IRQ_LEGACY);
      |                                     ^~~~~~~~~~~~~~
      |                                     NR_IRQS_LEGACY
aq_pci_func.c:383:37: note: each undeclared identifier is reported only once for each function it appears in
make[4]: *** [/usr/src/linux-headers-6.13.0-rc5-custom/scripts/Makefile.build:194: aq_pci_func.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.13.0-rc5-custom/Makefile:1989: .] Error 2
make[2]: *** [/usr/src/linux-headers-6.13.0-rc5-custom/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/opt/07-18-24_Marvell_Linux_2.5.12/Linux'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.13.0-rc5-custom'
make: *** [Makefile:36: all] Error 2

```

amlopin, 2025-01-15:
Hi I have got the same issue. Did you fix it? Thanks

`   CC [M]  /home/Linux/aq_main.o
  CC [M]  /home/Linux/aq_nic.o
  CC [M]  /home/Linux/aq_pci_func.o
  CC [M]  /home/Linux/aq_vec.o
/home/Linux/aq_pci_func.c: In function ‘aq_pci_probe’:
/home/Linux/aq_pci_func.c:383:37: error: ‘PCI_IRQ_LEGACY’ undeclared (first use in this function); did you mean ‘NR_IRQS_LEGACY’?
  383 |                                     PCI_IRQ_LEGACY);
      |                                     ^~~~~~~~~~~~~~
      |                                     NR_IRQS_LEGACY
/home/Linux/aq_pci_func.c:383:37: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [scripts/Makefile.build:229: /home/Linux/aq_pci_func.o] Error 1
make[3]: *** Waiting for unfinished jobs....
`

epox33d, 2025-02-15:
Stock kernel 6.1.0-31-amd64 SMP PREEMPT_DYNAMIC Debian 6.1.128-1.
Driver AQtion-2.5.12. Replaced file [aq_macsec.c](https://github.com/torvalds/linux/commit/27418b55f094c3c8bb125759338f6476ab70f043#diff-ade034eaf3344c443e0a1625577a411cddfce7da143bf8a8bac4b59897f3aa46) and it compiles without errors but i've got new issue:
**Skipping BTF generation for /opt/atlantic/atlantic.ko due to unavailability of vmlinux**

after make install:
**W: Possible missing firmware /lib/firmware/mrvl/91B1.fw for module atlantic
W: Possible missing firmware /lib/firmware/mrvl/87B1.fw for module atlantic
W: Possible missing firmware /lib/firmware/mrvl/80B1.fw for module atlantic**

Finally module doesn't works and here are no new ethernet apadters in `ip a` because of signature and/or required key missing (msg from dmesg):
**[Sat Feb 15 14:13:55 2025] atlantic: module verification failed: signature and/or required key missing - tainting kernel**

full output:
```
root@******:/opt/atlantic# make
make -j4 -C /lib/modules/6.1.0-31-amd64/build M="/opt/atlantic" modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-31-amd64'
  CC [M]  /opt/atlantic/aq_main.o
  CC [M]  /opt/atlantic/aq_nic.o
  CC [M]  /opt/atlantic/aq_pci_func.o
  CC [M]  /opt/atlantic/aq_vec.o
  CC [M]  /opt/atlantic/aq_ring.o
  CC [M]  /opt/atlantic/aq_ptp.o
  CC [M]  /opt/atlantic/aq_filters.o
  CC [M]  /opt/atlantic/aq_hw_utils.o
  CC [M]  /opt/atlantic/aq_ethtool.o
  CC [M]  /opt/atlantic/aq_drvinfo.o
  CC [M]  /opt/atlantic/aq_trace.o
  CC [M]  /opt/atlantic/aq_phy.o
  CC [M]  /opt/atlantic/aq_compat.o
  CC [M]  /opt/atlantic/aq_tsn.o
  CC [M]  /opt/atlantic/aq_sysfs.o
  CC [M]  /opt/atlantic/hw_atl/hw_atl_a0.o
  CC [M]  /opt/atlantic/hw_atl/hw_atl_b0.o
  CC [M]  /opt/atlantic/hw_atl/hw_atl_utils.o
  CC [M]  /opt/atlantic/hw_atl/hw_atl_utils_fw2x.o
  CC [M]  /opt/atlantic/hw_atl/hw_atl_llh.o
  CC [M]  /opt/atlantic/hw_atl/hw_atl_fw_hostboot.o
  CC [M]  /opt/atlantic/hw_atl/hw_atl_fw_image.o
  CC [M]  /opt/atlantic/hw_atl2/hw_atl2_llh.o
  CC [M]  /opt/atlantic/hw_atl2/hw_atl2.o
  CC [M]  /opt/atlantic/hw_atl2/hw_atl2_utils.o
  CC [M]  /opt/atlantic/hw_atl2/hw_atl2_utils_fw.o
  CC [M]  /opt/atlantic/hw_atl2/hw_atl2_fw_hostboot.o
  CC [M]  /opt/atlantic/macsec/macsec_api.o
  CC [M]  /opt/atlantic/aq_dash.o
  CC [M]  /opt/atlantic/aq_macsec.o
  LD [M]  /opt/atlantic/atlantic.o
  MODPOST /opt/atlantic/Module.symvers
  CC [M]  /opt/atlantic/atlantic.mod.o
  LD [M]  /opt/atlantic/atlantic.ko
  BTF [M] /opt/atlantic/atlantic.ko
Skipping BTF generation for /opt/atlantic/atlantic.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-31-amd64'
root@******:/opt/atlantic# modprobe -v atlantic
root@******:/opt/atlantic# rmmod atlantic
root@******:/opt/atlantic# make load
modprobe ptp
modprobe crc_itu_t
modprobe -q macsec || true
insmod ./atlantic.ko
root@aeronas:/opt/atlantic# make install
make[1]: Entering directory '/opt/atlantic'
atlantic.ko is in initramfs.
CAUTION! Updating initramfs is potentially dangerous.
Attempt initramfs update? [yN] y
update-initramfs: Generating /boot/initrd.img-6.1.0-31-amd64
W: Possible missing firmware /lib/firmware/mrvl/91B1.fw for module atlantic
W: Possible missing firmware /lib/firmware/mrvl/87B1.fw for module atlantic
W: Possible missing firmware /lib/firmware/mrvl/80B1.fw for module atlantic
make[1]: Leaving directory '/opt/atlantic'
root@******:/opt/atlantic# lsmod | grep atlantic
atlantic              344064  0
macsec                 61440  1 atlantic
crc_itu_t              16384  1 atlantic
root@******:/opt/atlantic# dmesg -T | grep -i atlantic
[Sat Feb 15 14:13:55 2025] atlantic: loading out-of-tree module taints kernel.
[Sat Feb 15 14:13:55 2025] atlantic: module verification failed: signature and/or required key missing - tainting kernel

```

The same stuff when using dkms:
```
root@******:/opt/atlantic# sudo ./dkms.sh install
Creating symlink /var/lib/dkms/atlantic/2.5.12.0/source -> /usr/src/atlantic-2.5.12.0
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.1.0-31-amd64 -C /lib/modules/6.1.0-31-amd64/build M=/var/lib/dkms/atlantic/2.5.12.0/build.....
Signing module /var/lib/dkms/atlantic/2.5.12.0/build/atlantic.ko
Cleaning build area...

atlantic.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.1.0-31-amd64/updates/dkms/
depmod...
root@******:/opt/atlantic# ls /lib/modules/6.1.0-31-amd64/updates/dkms/
atlantic.ko
root@******:/opt/atlantic# sudo ./dkms.sh install
Error! DKMS tree already contains: atlantic-2.5.12.0
You cannot add the same module/version combo more than once.
```

Rebuild kernel without 'Module signature verification' doesn't helps.
Edited kernel .config:
```
CONFIG_MODULE_SIG=n
CONFIG_MODULE_SIG_ALL=n
# CONFIG_MODULE_SIG_FORCE is not set 
# CONFIG_MODULE_SIG_SHA1 is not set
# CONFIG_MODULE_SIG_SHA224 is not set
# CONFIG_MODULE_SIG_SHA256 is not set
# CONFIG_MODULE_SIG_SHA384 is not set
```

whitefalk, 2025-03-05:
I was looking for the same issue with a differnet driver.
In Linux 6.10 PCI_IRQ_LEGACY was renamed to PCI_IRQ_INTX.
See: https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.10

rursache, 2025-03-05:
> I was looking for the same issue with a differnet driver. In Linux 6.10 PCI_IRQ_LEGACY was renamed to PCI_IRQ_INTX. See: https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.10

this and another 2 depreciations got fixed by claude 3.7 and I managed to both compile and run the driver, got the adapter working as well.

here is the fixed driver, follow the readme for install: 
[Marvell_Linux_2.5.12.zip](https://github.com/user-attachments/files/19092201/Marvell_Linux_2.5.12.zip)

nervtech, 2025-03-11:
I tested the fixed driver in Debian 12 (kernel 6.12.12) and found that I needed to modify the parameter type of the aq_ethtool_get_ts_info function in aq_ethtool.c to kernel_ethtool_ts_info for successful compilation. it's currently working fine.

I'm not entirely sure whether this modification will maintain stability with Debian 12's libraries.

exxocism, 2025-03-21:
I patched both `aq_ethtool_get_ts_info` function and `aq_sysfs.c` to make it work in ubuntu 24.04 (kernel 6.11.0).  USE AT YOUR OWN RISK.

[Marvell_Linux_2.5.12-patched-ubuntu-2404.zip](https://github.com/user-attachments/files/19391076/Marvell_Linux_2.5.12-patched-ubuntu-2404.zip)

rursache, 2025-06-05:
patched for Ubuntu 25.04 x86_64 and kernel 6.15.0

[Marvell_Linux_2.5.12-ubuntu-25.04_kernel-6.15.0.zip](https://github.com/user-attachments/files/20608211/Marvell_Linux_2.5.12-ubuntu-25.04_kernel-6.15.0.zip)

aidanc, 2025-09-13:
I'm trying to get this working with latest proxmox (9.0.9) but haven't had any success .. really wish Marvell would do better with these drivers..
Any chance someone could post a patched version compatible with 6.14.11-2-pve?

kishkaru, 2025-10-08:
Patched version, compliments of `claude-4.5-sonnet`.

Confirmed to compile, install, and tested working on kernel `6.17.1-arch1-1`

[Marvell_v2.5.12_kernel-6.17.1-Linux.tar.gz](https://github.com/user-attachments/files/22759498/Marvell_v2.5.12_kernel-6.17.1-Linux.tar.gz)

## Final Complete Summary of ALL Changes

### Files Modified:

 1. `aq_common.h`
 • Added `#include <linux/timer.h>` (line 19)
 2. `aq_compat.h`
 • Added `PCI_IRQ_*` flag definitions (lines 108-116)
 • Added `from_timer` macro fallback (lines 69-72)
 • Added `del_timer_sync` to `timer_delete_sync` mapping (lines 75-77)
 3. `aq_nic.c`
 • Added `#include "aq_common.h"` before other local includes (line 23)
 4. `aq_pci_func.c`
 • Added `#include "aq_common.h"` before other local includes (line 16)
 5. `aq_sysfs.c`
 • Added version-specific `memreg_mmap()` function signatures for `const` correctness (lines 266-276)
 6. `aq_ethtool.c`
 • Added version-specific `aq_ethtool_get_ts_info()` implementations for different `struct` types (lines 926-988)

### Issues Fixed:

✅ PCI_IRQ_LEGACY undeclared
✅ from_timer() implicit declaration
✅ del_timer_sync() implicit declaration / undefined symbol
✅ ethtool_ts_info incompatible pointer type
✅ bin_attribute mmap callback signature mismatch

The driver should now successfully compile and link on kernel 6.17.1 (Arch Linux)! 🎉

---

If you are using a Thunderbolt 3/4 or USB 4 device, don't forget to enable Thunderbolt authorization, and re-scan the PCIe bus after:

```
10:27 PM:~ $ sudo boltctl list
 ● Ubiquiti Inc. UACC-Adapter-RJ45-USBC-10GE
   ├─ type:          peripheral
   ├─ name:          UACC-Adapter-RJ45-USBC-10GE
   ├─ vendor:        Ubiquiti Inc.
   ├─ uuid:          c1010000-0072-7c1e-0392-1fe11a533909
   ├─ generation:    Thunderbolt 3
   ├─ status:        authorized     <--------------------------------------------------
   │  ├─ domain:     e5a43804-c1af-a3ca-ffff-ffffffffffff
   │  ├─ rx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  ├─ tx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  └─ authflags:  none
   ├─ authorized:    Wed 08 Oct 2025 05:27:05 AM UTC
   ├─ connected:     Wed 08 Oct 2025 05:22:04 AM UTC
   └─ stored:        Wed 08 Oct 2025 05:27:05 AM UTC
      ├─ policy:     iommu
      └─ key:        no

$ echo 1 | sudo tee /sys/bus/pci/rescan 
1

$ lspci -nn | grep "1d6a"
62:00.0 Ethernet controller [0200]: Aquantia Corp. AQtion AQC113 NBase-T/IEEE 802.3an Ethernet Controller [Antigua 10G] [1d6a:04c0] (rev 03)
```