Aquantia/AQtion #70: Wake on lan not working on linux with AQC113C

Slashic, 2025-02-12:
I tried different distros with different kernel versions, all behave the same.

Wake On Lan doesn't work and the NIC is shut down on Poweroff or Suspend.
This works perfectly fine on Windows 10 on the same hardware.

I also tried to use the driver provided on the official website, but they're not compiling as described in this issue: https://github.com/Aquantia/AQtion/issues/69

Here are the configurations on Fedora 41 KDE Spin:
Linux: 6.12.11-200.fc41.x86_64

Also tested on Linux Mint 22 & Ubuntu 24.04

```
>ethtool enp6s0
Settings for enp6s0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Full
                                100baseT/Full
                                1000baseT/Full
                                10000baseT/Full
                                2500baseT/Full
                                5000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Full
                                100baseT/Full
                                1000baseT/Full
                                10000baseT/Full
                                2500baseT/Full
                                5000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Full
                                             100baseT/Full
                                             1000baseT/Full
                                             10000baseT/Full
                                             2500baseT/Full
                                             5000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 10000Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        MDI-X: Unknown
        Supports Wake-on: pg
        Wake-on: g
        Current message level: 0x00000005 (5)
                               drv link
        Link detected: yes

```

ACPI:

```
>cat /proc/acpi/wakeup
Device  S-state   Status   Sysfs node
PEG1      S4    *enabled   pci:0000:00:01.0
PEGP      S4    *enabled   pci:0000:01:00.0
PEG2      S4    *disabled
PEGP      S4    *disabled
PEG0      S4    *enabled   pci:0000:00:06.0
PEGP      S4    *disabled  pci:0000:04:00.0
RP09      S4    *disabled  pci:0000:00:1d.0
PXSX      S4    *disabled
RP10      S4    *disabled
PXSX      S4    *disabled
RP11      S4    *disabled
PXSX      S4    *disabled
RP12      S4    *disabled
PXSX      S4    *disabled
RP13      S4    *enabled   pci:0000:00:1d.4
PXSX      S4    *disabled  pci:0000:08:00.0
RP14      S4    *disabled
PXSX      S4    *disabled
RP15      S4    *disabled
PXSX      S4    *disabled
RP16      S4    *disabled
PXSX      S4    *disabled
RP01      S4    *enabled   pci:0000:00:1c.0
PXSX      S4    *enabled  pci:0000:06:00.0
RP02      S4    *disabled
PXSX      S4    *disabled
RP03      S4    *disabled
PXSX      S4    *disabled
RP04      S4    *disabled
PXSX      S4    *disabled
RP05      S4    *disabled
PXSX      S4    *disabled
RP06      S4    *disabled
PXSX      S4    *disabled
RP07      S4    *disabled
PXSX      S4    *disabled
RP08      S4    *disabled
PXSX      S4    *disabled
RP17      S4    *disabled
PXSX      S4    *disabled
RP18      S4    *disabled
PXSX      S4    *disabled
RP19      S4    *disabled
PXSX      S4    *disabled
RP20      S4    *disabled
PXSX      S4    *disabled
RP21      S4    *disabled
PXSX      S4    *disabled
RP22      S4    *disabled
PXSX      S4    *disabled
RP23      S4    *disabled
PXSX      S4    *disabled
RP24      S4    *disabled
PXSX      S4    *disabled
RP25      S4    *enabled   pci:0000:00:1a.0
PXSX      S4    *disabled  pci:0000:05:00.0
RP26      S4    *disabled
PXSX      S4    *disabled
RP27      S4    *disabled
PXSX      S4    *disabled
RP28      S4    *disabled
PXSX      S4    *disabled
XHCI      S4    *enabled   pci:0000:00:14.0
XDCI      S4    *disabled
HDAS      S4    *disabled  pci:0000:00:1f.3
CNVW      S4    *disabled  pci:0000:00:14.3
AWAC      S4    *enabled   platform:ACPI000E:00
```

Infos:
```
>ethtool -i enp6s0
driver: atlantic
version: 6.12.11-200.fc41.x86_64
firmware-version: 1.3.7
expansion-rom-version: 
bus-info: 0000:06:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes

```

epox33d, 2025-02-15:
1. What vesrsion of AQtion driver do you use?
2. How did you solve issue "atlantic: module verification failed: signature and/or required key missing - tainting kernel"?

Slashic, 2025-02-15:
About the driver version, these are embedded ones in the kernel itself, i suppose it shows the kernel version instead of the standalone one you normally have here.

I didn't compile the driver as i was stuck on that error, but i see you managed to find some c files to get a build alas not working.

epox33d, 2025-02-15:
Can you show your `modinfo atlantic`, `lsmod | grep atlantic` and `lspci -nn | grep 1d6a:04c0` output? Just like this:

```
root@******:~# modinfo atlantic
filename:       /lib/modules/6.1.0-31-amd64/updates/drivers/net/ethernet/aquantia/atlantic/atlantic.ko
description:    Marvell (Aquantia) Corporation(R) Network Driver
author:         Marvell
version:        2.5.12.0
license:        GPL v2
firmware:       mrvl/91B1.fw
firmware:       mrvl/87B1.fw
firmware:       mrvl/80B1.fw
srcversion:     257DE98845AFD5081AC22C8
alias:          pci:v00001D6Ad000003C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000093C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000094C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000011C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000034C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000014C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000004C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000092B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000091B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000089B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000088B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000087B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000080B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000011B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000009B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000008B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000007B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D109sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D108sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D107sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D100sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad00000001sv*sd*bc*sc*i*
depends:        macsec,crc-itu-t
retpoline:      Y
name:           atlantic
vermagic:       6.1.0-31-amd64 SMP preempt mod_unload modversions
parm:           aq_ptp_offset_forced:Force to use the driver parameters (uint)
parm:           aq_ptp_offset_100:PTP offset for 100M (uint)
parm:           aq_ptp_offset_1000:PTP offset for 1G (uint)
parm:           aq_ptp_offset_2500:PTP offset for 2,5G (uint)
parm:           aq_ptp_offset_5000:PTP offset for 5G (uint)
parm:           aq_ptp_offset_10000:PTP offset for 10G (uint)
parm:           aq_ptp_gpio_hightime:PTP GPIO high time (uint)
parm:           sleep_delay:uint
parm:           aq_itr:Interrupt throttling mode (uint)
parm:           aq_itr_tx:TX interrupt throttle rate (uint)
parm:           aq_itr_rx:RX interrupt throttle rate (uint)
parm:           aq_rxpageorder:RX page order override (uint)
parm:           aq_rx_refill_thres:RX refill threshold (uint)
parm:           debug:Default debug msglevel (uint)
parm:           aq_fw_did:Use FW image for this DID (array of uint)
parm:           aq_fw_sid:Use provisioning data for this SID (array of uint)
parm:           aq_force_host_boot:Force host boot (array of uint)
parm:           aq_enable_wa:Quirk bits to enable HW workarounds (int)
parm:           aq_enable_ptp:Enable PTP (bool)
root@******:~# lsmod | grep atlantic
atlantic              344064  0
macsec                 61440  1 atlantic
crc_itu_t              16384  1 atlantic
root@******:~# lspci -nn | grep 1d6a:04c0
root@******:~#

```

Slashic, 2025-02-18:
Here they are:

```
filename:       /lib/modules/6.12.13-200.fc41.x86_64/kernel/drivers/net/ethernet/aquantia/atlantic/atlantic.ko.xz
description:    Marvell (Aquantia) Corporation(R) Network Driver
author:         Marvell
license:        GPL v2
alias:          pci:v00001D6Ad000011C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000034C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000014C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000004C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000093C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000094C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000C0sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000092B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000091B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000089B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000088B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000087B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000080B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000012B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000011B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000009B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000008B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000007B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad000000B1sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D109sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D108sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D107sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad0000D100sv*sd*bc*sc*i*
alias:          pci:v00001D6Ad00000001sv*sd*bc*sc*i*
depends:        macsec
intree:         Y
name:           atlantic
retpoline:      Y
vermagic:       6.12.13-200.fc41.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:       
sig_hashalgo:   sha256
signature:      
parm:           aq_itr:Interrupt throttling mode (uint)
parm:           aq_itr_tx:TX interrupt throttle rate (uint)
parm:           aq_itr_rx:RX interrupt throttle rate (uint)
```

```
atlantic              344064  0
macsec                 73728  1 atlantic
```

```
lspci -nn | grep 06:00.0
06:00.0 Ethernet controller [0200]: Aquantia Corp. AQC113C NBase-T/IEEE 802.3an Ethernet Controller [Marvell Scalable mGig] [1d6a:14c0] (rev 03)
```

Slashic, 2025-03-05:
No solution yet, i tried various distros and kernel version, they all have that same embedded driver version that is i believe taken from these and fixed by kernel maintainers.
I gave up on wake on lan with Marvell NICs, i spent too much time on this already.

zeroepoch, 2025-06-11:
WOL doesn't appear to work with the upstream driver. I suppose there is some bug there. When I used the official Marvell driver, which needs patches mentioned in https://github.com/Aquantia/AQtion/issues/69, it started working right away.

Bobo1239, 2025-06-14:
Can confirm that the patch applied to the current Arch kernel (`6.15.2.arch1-1`) also works on my hardware (GIGABYTE B850 AI TOP with dual AQC113C). Thank you! 👍

@zeroepoch Do you intend to upstream your patch? If not me or someone else could also take care of that part.

zeroepoch, 2025-06-15:
Yep, I'm going to try and start the upstream process next week. I have a friend who has worked on other network changes upstream who is going to help me through the process.

I also tried this on a NICGIGA AQC113C (https://www.amazon.com/dp/B0D87439N7) and it worked as well. I previously made the change on a TP-Link TX401 which has a AQC113, FYI.

zeroepoch, 2025-06-15:
I case anyone comes across this issue who is using Fedora, I've created some akmod packages that build the patched `atlantic` driver. With akmod packages it will rebuild the kernel module each time you update the kernel. At least until the snapshot I took of the upstream Linux kernel is incompatible, at which point I'll need to refresh the snapshot. As others suggested I would like to get this patch upstreamed, but this is my solution until then, and it was something I wanted to play with at some point anyway (akmods and Copr).

Links:
- My Fedora Copr project: https://copr.fedorainfracloud.org/coprs/ework/atlantic-kmod/
  - Package source: https://github.com/zeroepoch/atlantic-kmod
- Patch in my fork of the kernel: https://github.com/zeroepoch/net-next/commit/3a01ee5f80233feb9bce2a39526a59e8f2d9f6d5

zeroepoch, 2025-07-27:
> So hopefully there is some way just to get UNRAID to update their kernel device drivers for this particular chipset.

If you're unfamiliar with how to compile a kernel module there might not be much other options right now. I believe my patch may land in kernel 6.17, so the other option is wait until that kernel version is released and UNRAID picks it up. That could be awhile though.

zeroepoch, 2025-08-10:
My patch is included in 6.17-rc1 so this issue should be resolved once the 6.17 kernel starts shipping.